balena-io/etcher #3818: usb drive randomly stop being able to flashed

maxtraxv3, 2022-10-02:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** windows 11 22h2 update
- **Image flashed:** ubuntu, 22.04 ubuntu server 22.04, fedora 36, cent os 9, void, antix
- **What do you think should have happened:**  flashed the usb.
- **What happened:** error? this has happend 2 usb driver, but they formant and work fine, after a few flashes etcher become unable to flash that usb drive? why i dont know, they work fine.
- **Do you see any meaningful error information in the DevTools?** 

Mon Oct 03 2022 04:04:52 GMT+1000 (Australian Eastern Standard Time) SafeWebview loaded ({"event":{"id":138,"url":"https://www.balena.io/etcher/success-banner/?borderTop=false&darkBackground=true&etcher-version=1.7.9&api-version=2&optOutAnalytics=true","method":"GET","timestamp":1664733892536.041,"resourceType":"mainFrame","ip":"54.206.202.192","fromCache":false,"statusLine":"HTTP/1.1 200","statusCode":200,"responseHeaders":{"age":["44308"],"cache-control":["public, max-age=0, must-revalidate"],"content-encoding":["br"],"content-length":["6743"],"content-type":["text/html; charset=UTF-8"],"date":["Sun, 02 Oct 2022 05:46:27 GMT"],"etag":["\"dd9bdb0990e5622131d7228ea4812cbf-ssl-df\""],"link":["</webpack-runtime-e524511d0b4a968a7b88.js>; rel=preload; as=script, </styles-c4fa86dfd4ad05097eee.js>; rel=preload; as=script, </framework-004503f2df4bb68d43f8.js>; rel=preload; as=script, </50c25a04-a68e3217d6ce65727bd7.js>; rel=preload; as=script, </c0d53ec4-df4e584640768dd709a2.js>; rel=preload; as=script, </532a2f07-f1ad28da1825cfcce2c7.js>; rel=preload; as=script, </app-c909f5221fe55e6f7849.js>; rel=preload; as=script, </c943faba-f859f80ceb7f4f7eab31.js>; rel=preload; as=script, </af13d906-36fdbc23fc7dd2f2b2e7.js>; rel=preload; as=script, </ec1189df-4ae9e03f4545d4f1d61b.js>; rel=preload; as=script, </a9a7754c-29508ff4045f1f93a5f6.js>; rel=preload; as=script, </commons-cea6bfcc846ea7ff6166.js>; rel=preload; as=script, </c974efff6ac245e04f4a26711b42db767846e0e5-b1b7091203b51af325ff.js>; rel=preload; as=script, </component---src-pages-etcher-success-banner-js-98ac1a1084e1d2a478fa.js>; rel=preload; as=script, </page-data/app-data.json>; rel=preload; as=fetch; crossorigin, </page-data/etcher/success-banner/page-data.json>; rel=preload; as=fetch; crossorigin"],"referrer-policy":["strict-origin-when-cross-origin"],"server":["Netlify"],"strict-transport-security":["max-age=31536000"],"vary":["Accept-Encoding"],"x-content-type-options":["nosniff"],"x-frame-options":["DENY"],"x-nf-request-id":["01GECXYPC0GNWQHHPKJJT1XXRH"],"x-xss-protection":["1; mode=block"]},"referrer":"","error":"net::OK"},"sample":0.1,"applicationSessionUuid":"3c45cb75-ae77-4fa3-9e49-5bfe4ec2649b","flashingWorkflowUuid":"f4daa853-1860-40ec-a18c-ddac6c6c5013"})
gui.js:46 Message from SafeWebview: Error: Not supported
gui.js:46 Mon Oct 03 2022 04:04:57 GMT+1000 (Australian Eastern Standard Time) Open external link ({"url":"https://github.com/balena-io/etcher/blob/master/CHANGELOG.md","sample":0.1,"applicationSessionUuid":"3c45cb75-ae77-4fa3-9e49-5bfe4ec2649b","flashingWorkflowUuid":"f4daa853-1860-40ec-a18c-ddac6c6c5013"})

this was me trying to flash to a usb that had been flash by it 2 time before, this i was using Ubuntu desktop 22.04

normal user, pluged into usb3 front port, programs open VLC, vivaldi (chrome based web browser), steam, discord.

mcraa, 2022-10-03:
Hi,

thanks for the info, the randomness is indeed interesting.
Can you try the failing part manually with the same device?

> Open `cmd.exe`
Execute `diskpart.exe`, it gives you diskpart's command line prompt
Execute `list disk` inside diskpart to show the available drives. Take note of the `id` of the drive in question
Execute `select disk N`, where `N` is the id of the removable drive obtained above (make sure you select the right one, otherwise you might wipe out another drive unintentionally)
Execute `clean`
check the result then exit from diskpart and cmd

Does it give any error or it cleans the drive?

maxtraxv3, 2022-10-03:
no it does not, works fine, on all 3 usb drives that are all different brands, one is usb 2 and the others usb 3.

laymanstake, 2022-10-27:
The same issue broken two of my usb drives in attempts to flash centos images while rufus can do the same and so is dd.  OS is Windows 11 22H2. Version is latest

closed 2023-07-12 by dfunckt