balena-io/etcher #3822: issue with bz2 compression

johanlantz1, 2022-10-17:
- Etcher version:1.7.9 
- Operating system and architecture:win10
- Image flashed:wic.bz2 
- What do you think should have happened:cannot handle bz2 compression.
- What happened:do not boot main image.

lurch, 2022-10-17:
Sounds similar to #3138 ?

johanlantz1, 2022-10-17:
Yes, and now 2 years later is this still an issue?

PatheticCode, 2023-02-08:
I am also running into troubles with .wic.bz2 files
    Etcher version: 1.14.3
    Operating system and architecture: linux, x86-64
    Image flashed:wic.bz2 (>8gb)
    What do you think should have happened: flashed sd-card
    What happened: etcher starts hanging at 5% and crashes after a while. 
The .wic.bz2 file is generated with yocto using the raspberrypi layer

mbulut, 2024-06-06:
Same here in 2024...

moto-timo, 2024-11-21:
With 1.19.25, I am seeing bz2 decompression stall at 63%. When I use bzunzip2 on the command line there are no issues with the extracted wic file. I had no problems with an earlier version, but I do not remember what the version was (probably installed within the last year).

HerrMuellerluedenscheid, 2025-04-11:
Hi, same issue on windows with bz2 stuck at 20%