balena-io/etcher #3823: macOS Error - EPERM: operation not permitted

ram-on, 2022-10-23:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** macOS 12.6
- **Image flashed:** ubuntu-22.10-desktop-amd64.iso
- **What do you think should have happened:** Image should have been flashed.
- **What happened:** After selecting the ISO image and the USB flash drive, Etcher ask me to input my password, which I did.  A second later, Etcher displays "Flash Failed!" and the outputted reason is `EPERM: operation not permitted, open '/dev/rdisk3'`
- **Do you see any meaningful error information in the DevTools?** 
```
"error": {
		"name": "Error",
		"message": "EPERM: operation not permitted, open '/dev/rdisk3'",
		"stack": "Error: EPERM: operation not permitted, open '/dev/rdisk3'",
		"code": "EPERM",
		"syscall": "open",
		"errno": -1
	},
```

I've tried the above process using different USB pendrives and different USB ports: all lead to the same outcome.

[image]
[image]

mobileteamz, 2022-11-24:
I am seeing the exact same thing in MacOS Monterey 12.3.1

manfredsteger, 2023-12-13:
same here 13.5 (22G74) Ventura - same Build on Win 11 works 😭

lurch, 2023-12-18:
I wonder if this might be the same thing as #3369 ?

seanPhill, 2024-08-18:
Who knows, but having had this happen, and I have a three year old version of the software, the first thing I'm going to do is upgrade from Etcher version 1.5.115 to the current v1.19.22. 🤞🏻
[image]

seanPhill, 2024-08-18:
:( Well, it still failed. This time I only got an error in my notifications, and the app itself was just grey and blank.
[image]

Now, I am wondering if it is something to do with trying to clone a running system drive 500 GB PCIe M.2 SSD to a 1000 GB PCIe M.2 SSD to update my 2013 Mac Pro! I don't have another system now, to run both as external drives.
[image]

seanPhill, 2024-08-18:
One further piece of information! I am using APFS on the source drive, and the target drive is not yet formatted.

seanPhill, 2024-08-18:
Update. Whether Balena Etcher can deal with cloning the System disk or not, I was able to do it by starting the Mac in Recovery Mode (⌘R + the power button) and using Disk Utility there, to clone one drive to the other.
Etcher may not have catered for this scenario, and you should be able to do such a clone using Etcher if you have another Mac with Etcher on it, and connect the computers with the one you want to clone booted in Target Disk  mode (⌘T before startup) and the external drive also connected. However at this point you might as well do it the way I did.
Balena could do this if they have an alternative Boot method that makes Etcher available, and if they provide some guidance in the software when you select a Boot Volume to be cloned.