balena-io/etcher #3826: CLI Interface?

ShaneMurphy2, 2022-10-25:
This tool has a lot of advantages of something like `dd`, but doesn't fit well into automation. Any plans for a CLI?

mcraa, 2022-10-26:
You can start etcher with command line parameters, but it does not really help, still opens up the GUI.
You can also directly run the [SDK](https://github.com/balena-io-modules/etcher-sdk/blob/master/examples/multi-destination.ts) but it is not designed to be a CLI.
There is some functionality exposed also in the [balena CLI](https://www.balena.io/docs/reference/balena-cli/#local-flash-image)

So the building blocks are there, we need to make it usable and user-friendly.
It is further down in our backlog.

lurch, 2022-10-26:
Rleated (still open) issues: #3483 #2381 #2363 #2274 #2742

ShaneMurphy2, 2022-10-26:
Thanks for the info!

zearp, 2023-08-24:
They stopped including the CLI in version after 1.4.9 -- I still use 1.4.9 cli to this day with no issues whatsoever. there was no reason to ever stop shipping it other than to get more impressions for their "ads" inside the gui.

https://github.com/balena-io/etcher/releases?q=1.4.9&expanded=true

closed 2025-10-06 by ShaneMurphy2