balena-io/etcher #3827: Feature: Be able to black list some drives

TomasNielsen, 2022-10-26:
I'm always scared to flash my Dymo printer och photo SD card by mistake. Would it be possible to black list certain drives that should never be flashed?

lurch, 2022-10-26:
Duplicate of #3147 but I've no idea if the feature described there still works? :shrug:

TomasNielsen, 2022-10-28:
The file does not exist any more. I tried changing the preferences file in the same manner but my changes are removed at every start of Balena and blackList does not work.
It would be totally fine with a settings file. Pretty please?

mcraa, 2022-10-28:
If the config.json file does not exist, you can create one, pay attention to use LF line endings.
The feature is not removed. => https://github.com/balena-io/etcher/blob/5945ab1f5096d230c4bd2ff912a7a3fd677ae2fb/lib/gui/app/app.ts#L173
The examples from [drivelist](https://github.com/balena-io-modules/drivelist) help what strings you may put there.
But it is a good idea to have the possibility somewhere on the UI.