balena-io/etcher #3828: Unable to launch both appimage and dnf repo

NicTanghe, 2022-10-30:
- **Etcher version:** 
- **Operating system and architecture:** 
- fedora 36
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?** 
- 
❯ ./balenaEtcher-1.7.9-x64.AppImage
[Axios v1.7.9] Transitional option 'clarifyTimeoutError' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'forcedJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
[Axios v1.7.9] Transitional option 'silentJSONParsing' has been deprecated since v1.0.0 and will be removed in the near future
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
libva error: vaGetDriverNameByIndex() failed with unknown libva error, driver_name = (null)
[10404:1030/174845.410379:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/tmp/.mount_balenafWvEty/balena-etcher-electron: line 10: 10404 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher-electron.bin "$@"

mcraa, 2022-10-31:
duplicate of [many](https://github.com/balena-io/etcher/search?q=GPU+process+usable&type=issues)
comment [here](https://github.com/balena-io/etcher/issues/3796#issuecomment-1250067751) could be most helpful

closed 2022-10-31 by mcraa

NicTanghe, 2022-10-31:
hello sorry that i did not hit those other issues,
On the other hand if so manny people report this as a bug.
maybe its verry stupid not to have 
/usr/bin/balena-etcher-electron --disable-gpu-sandbox
as the default and --enable-gpu-sandbox as an option