balena-io/etcher #3873: Unable to install on Ubuntu 22.10

philclifford, 2022-11-11:
- **Etcher version:** 
- 1.81 1.9.0 and 1.10.0
- **Operating system and architecture:** 
- Ubuntu Mate 22.10 amd64
- OS: Ubuntu MATE 22.10 x86_64 
Host: Latitude E6530 01 
Kernel: 5.19.0-23-generic 
Uptime: 7 hours, 51 mins 
Packages: 4827 (dpkg), 25 (flatpak), 40 (snap) 
Shell: bash 5.2.2 
Resolution: 1600x900 
DE: MATE 1.26.0 
WM: Metacity (Marco) 
Theme: Yaru-MATE-Pink-dark [GTK2/3] 
Icons: Yaru-MATE-Pink-dark [GTK2/3] 
Terminal: terminus 
CPU: Intel i7-3540M (4) @ 3.700GHz 
GPU: Intel 3rd Gen Core processor Graphics Controller 
GPU: NVIDIA NVS 5200M 
Memory: 4702MiB / 7820MiB 

- **Image flashed:** 
- n/a
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?** 

Desktop Launch silently fails.  typing `balena-etcher`in a shell gets the following error output:

```bash
[200744:1111/195746.053511:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
/usr/bin/balena-etcher: line 10: 200744 Trace/breakpoint trap   (core dumped) "${script_dir}"/balena-etcher.bin "$@"
```
Just trying to test update/upgrade using the github releases via `deb-get` and found it is now broken on 22.10.
Incidentally the attempt to update in place via apt-get (from 1.7.x ) was failing because of packaging conflicts (I'm guessing a naming change led to the complaint that was something about refusing to overwrite a library that was "also owned by" the balena-etcher being upgraded ???) but then appeared to work OK, but left this non-working state.

philclifford, 2022-11-11:
Also same error from the 1.7.9 AppImage

lurch, 2022-11-13:
Duplicate of many other issues https://github.com/search?q=repo%3Abalena-io%2Fetcher+gpu+process+usable&type=issues

crowetic, 2022-11-14:
The way you can temporarily get around this issue, is to open the folder the AppImage is in in a terminal, then run the command like this...

```./balenaEtcher-1.7.9-x64.AppImage --no-sandbox```

Hopefully the balena devs will get around to resolving this, as I agree it's an unnecessary step and rather annoying to deal with.

But this works for me, and the application runs perfectly once you do this. Just be sure you DO NOT CLOSE THE TERMINAL you used to launch the app, until you're done with the app, as this will cause the app to close, since it was launched with the terminal.

philclifford, 2022-11-17:
Thanks @crowetic.  Helpful info for anyone wanting to run the AppImage.

LiamDawe, 2022-11-17:
This is also a problem on Fedora 36

`--no-sandbox` works around it too, quite annoying

milespossing, 2022-12-12:
Confirming for pop os

`--no-sandbox` is key

gagaltotal, 2022-12-16:
zorin os not working --no-sandbox
version balenaEtcher-1.10.2-x64.AppImage

```
./balenaEtcher-1.10.2-x64.AppImage --no-sandbox
mesa: for the --simplifycfg-sink-common option: may only occur zero or one times!
mesa: for the --global-isel-abort option: may only occur zero or one times!
mesa: for the --amdgpu-atomic-optimizations option: may only occur zero or one times!
ready-to-show: 905.005ms
```

dfunckt, 2023-07-12:
Closing in favor of #4023

closed 2023-07-12 by dfunckt