balena-io/etcher #3874: Etcher overwrote both my USB stick and my SD card, but I told it to only write to my SD card

nicco1690, 2022-11-13:
- **Etcher version:** 1.7.9, portable
- **Operating system and architecture:** Windows 10, x86_64
- **Image flashed:** A combined and extracted version of https://github.com/eupnea-linux/eupnea-os/releases, as stated in its instructions
- **What do you think should have happened:** Etcher should've only wrote to my SD card
- **What happened:** Etcher wrote to my USB stick and my SD card, erasing the data on my USB stick.
- **Do you see any meaningful error information in the DevTools?** This is not applicable, as I've already closed and deleted the app from my system.

I've done this as I don't plan on ever using it again due to this issue. If an app has an issue as trivial to fix as this yet as critically important that it doesn't happen, then said app should not be used, especially not in a case where it is critical that said data should be kept.

In addition, a warning dialog should be added when you press the write button that confirms that the choices that you have entered are correct, and inform the user that writing to that drive will likely delete the data on it if there is data on it, like how Rufus does it.

crowetic, 2022-11-14:
Can't say I have ever seen such a thing before, and I have used etcher with multiple operating systems, mutiple drives, countless times... I have a really hard time believing you didn't accidentally select both drives or something. I highly doubt a bug of this magnitide would have stayed unnoticed for that long.

But I'll let the Balena devs chime in on this one. 

Good luck finding a better imaging tool than etcher.

aethernet, 2022-11-25:
As @crowetic just said, it's very very very unlikely that it's a bug and not a user mistake.

We'd love to get a repro if you can.

That being said, it might make sense to add a warning where multiple devices are selected, to avoid such mistake ?

closed 2022-12-02 by aethernet