balena-io/etcher #3876: Balena Etcher won't start

mapagm, 2022-11-17:
- **Etcher version:** 1.7.9
- **Operating system and architecture:** KDE Neon 22.04 x64
- **Image flashed:** -

Hi. I just installed Balena Etcher via repository as shown here: https://cloudsmith.io/~balena/repos/etcher/setup/#formats-deb. It seems it installed correctly - i have balena icon in menu:
[image]

But when I try to start it, the icon just flash for a second and nothing happens...

ab77, 2022-11-17:
@mcraa @zwhitchcox folks, I feel like we've not got a handle on Etcher distribution... Do you know what process publishes to cloudsmith.io?

The source of truth for Etcher is https://www.balena.io/etcher/, which fetches (out of date) release from GitHub. We've also had community maintainers pull support for Homebrew, AUR, etc., since our release cadence increased.

I feel like we need a single source of truth for Etcher releases (official) and if we want to support additional package repositories, we should automate it in our CI pipeline and not rely for external contributors for distribution (core piece of Etcher).

cc @chrisys

cheeeeee, 2022-11-18:
Try putting `balena-etcher-electron` in a terminal directly. I'm getting `/usr/lib/electron12/electron: error while loading shared libraries: libicui18n.so.71: cannot open shared object file: No such file or directory` . Haven't solved it yet. Appears to be a dependency issue.

hillz2, 2022-12-04:
> 

It still happens to me, has anyone fixed it yet ?

mcraa, 2022-12-29:
https://github.com/balena-io/etcher/issues/3771#issuecomment-1155239643

closed 2023-07-12 by dfunckt