balena-io/etcher #3883: Failed to build from source on archlinux

sukanka, 2022-11-22:
Hello, I'm one of the aur maintainers of [balena-etcher](https://aur.archlinux.org/packages/balena-etcher).
Currently, I'm planning to build it from source on archlinux following the arch way.

[Here](https://gist.github.com/sukanka/792fc4b100294ff7a1317e6c04c87302#file-pkgbuild) is my PKGBUILD. But I failed to build this app (more specifically, the module `/build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io`),  and the logs are shown below

```
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'balena-etcher@1.10.1',
npm WARN EBADENGINE   required: { node: '>=14 < 16' },
npm WARN EBADENGINE   current: { node: 'v18.12.1', npm: '8.19.2' }
npm WARN EBADENGINE }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: 'etcher-sdk@7.4.0',
npm WARN EBADENGINE   required: { node: '>=14 < 18' },
npm WARN EBADENGINE   current: { node: 'v18.12.1', npm: '8.19.2' }
npm WARN EBADENGINE }
npm WARN EBADENGINE Unsupported engine {
npm WARN EBADENGINE   package: '@balena/node-beaglebone-usbboot@2.0.1',
npm WARN EBADENGINE   required: { node: '>=14 < 18' },
npm WARN EBADENGINE   current: { node: 'v18.12.1', npm: '8.19.2' }
npm WARN EBADENGINE }
npm WARN deprecated querystring@0.2.0: The querystring API is considered Legacy. new code should use the URLSearchParams API instead.
npm WARN deprecated @types/puppeteer@7.0.4: This is a stub types definition. puppeteer provides its own type definitions, so you do not need this installed.
npm WARN deprecated stable@0.1.8: Modern JS already guarantees Array#sort() is a stable sort, so this library is deprecated. See the compatibility table on MDN: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/sort#browser_compatibility
npm WARN deprecated raven@2.6.4: Please upgrade to @sentry/node. See the migration guide https://bit.ly/3ybOlo7
npm WARN deprecated raven-js@3.27.2: Please upgrade to @sentry/browser. See the migration guide https://bit.ly/3ybOlo7
npm WARN deprecated @braintree/sanitize-url@3.1.0: Potential XSS vulnerability patched in v6.0.0.
npm WARN deprecated core-js@2.6.12: core-js@<3.23.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Some versions have web compatibility issues. Please, upgrade your dependencies to the actual version of core-js.
npm WARN deprecated @types/ajv-keywords@3.5.0: This is a stub types definition. ajv-keywords provides its own type definitions, so you do not need this installed.
npm WARN deprecated @npmcli/move-file@2.0.1: This functionality has been moved to @npmcli/fs
npm WARN deprecated asar@3.2.0: Please use @electron/asar moving forward.  There is no API change, just a package name change
npm WARN deprecated svgo@1.3.2: This SVGO version is no longer supported. Upgrade to v2.x.x.
npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm WARN deprecated tslint@6.1.3: TSLint has been deprecated in favor of ESLint. Please see https://github.com/palantir/tslint/issues/4534 for more information.
npm WARN deprecated electron-rebuild@3.2.9: Please use @electron/rebuild moving forward.  There is no API change, just a package name change
npm WARN deprecated electron-notarize@1.0.0: Please use @electron/notarize moving forward.  There is no API change, just a package name change
npm WARN deprecated debug@4.2.0: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm WARN deprecated uuid@3.3.2: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated core-js@1.2.7: core-js@<3.23.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Some versions have web compatibility issues. Please, upgrade your dependencies to the actual version of core-js.
npm WARN deprecated coffee-script@1.11.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
npm WARN deprecated debug@3.2.6: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
npm notice 
npm notice New major version of npm available! 8.19.2 -> 9.1.2
npm notice Changelog: <https://github.com/npm/cli/releases/tag/v9.1.2>
npm notice Run `npm install -g npm@9.1.2` to update!
npm notice 
npm ERR! code 1
npm ERR! path /build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io
npm ERR! command failed
npm ERR! command sh -c -- node-gyp rebuild
npm ERR! make: Entering directory '/build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io/build'
npm ERR!   CC(target) Release/obj.target/binding/binding.o
npm ERR! make: Leaving directory '/build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io/build'
npm ERR! gyp info it worked if it ends with ok
npm ERR! gyp info using node-gyp@9.3.0
npm ERR! gyp info using node@18.12.1 | linux | x64
npm ERR! gyp info find Python using Python version 3.10.8 found at "/usr/bin/python3"
npm ERR! gyp http GET https://nodejs.org/download/release/v18.12.1/node-v18.12.1-headers.tar.gz
npm ERR! gyp http 200 https://nodejs.org/download/release/v18.12.1/node-v18.12.1-headers.tar.gz
npm ERR! gyp http GET https://nodejs.org/download/release/v18.12.1/SHASUMS256.txt
npm ERR! gyp http 200 https://nodejs.org/download/release/v18.12.1/SHASUMS256.txt
npm ERR! gyp info spawn /usr/bin/python3
npm ERR! gyp info spawn args [
npm ERR! gyp info spawn args   '/build/balena-etcher/src/etcher-1.10.1/node_modules/node-gyp/gyp/gyp_main.py',
npm ERR! gyp info spawn args   'binding.gyp',
npm ERR! gyp info spawn args   '-f',
npm ERR! gyp info spawn args   'make',
npm ERR! gyp info spawn args   '-I',
npm ERR! gyp info spawn args   '/build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io/build/config.gypi',
[... 20 lines cut ...]
npm ERR! make: *** Waiting for unfinished jobs....
npm ERR! gyp ERR! build error 
npm ERR! gyp ERR! stack Error: `make` failed with exit code: 2
npm ERR! gyp ERR! stack     at ChildProcess.onExit (/build/balena-etcher/src/etcher-1.10.1/node_modules/node-gyp/lib/build.js:203:23)
npm ERR! gyp ERR! stack     at ChildProcess.emit (node:events:513:28)
npm ERR! gyp ERR! stack     at ChildProcess._handle.onexit (node:internal/child_process:291:12)
npm ERR! gyp ERR! System Linux 6.0.9-zen1-1-zen
npm ERR! gyp ERR! command "/usr/bin/node" "/build/balena-etcher/src/etcher-1.10.1/node_modules/.bin/node-gyp" "rebuild"
npm ERR! gyp ERR! cwd /build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io
npm ERR! gyp ERR! node -v v18.12.1
npm ERR! gyp ERR! node-gyp -v v9.3.0
npm ERR! gyp ERR! not ok

npm ERR! A complete log of this run can be found in:
npm ERR!     /build/balena-etcher/src/.npm/_logs/2022-11-22T12_48_20_359Z-debug-0.log
[1m[31m==> ERROR:(B[m[1m A failure occurred in build().(B[m
[1m    Aborting...(B[m
```

- **Etcher version:** 1.10.1
- **Operating system and architecture:**  archlinux  Kernel: 6.0.9-zen1-1-zen  x86_64
- **Image flashed:** None
- **What do you think should have happened:**  build the app successfully
- **What happened:** The build process failed
- **Do you see any meaningful error information in the DevTools?** No

killi1812, 2023-01-11:
Hi, I was wandering Is there any progress or work around on this problem

mcraa, 2023-01-11:
Hi, the direct-io package was updated 3 years ago last time, it caused us problems many times as well, we are working on a replacement.

clarencefoy, 2023-03-03:
I'd like to mention that this appears to be occurring on Manjaro as of 03-03-2023 as well.

zhullyb, 2023-04-07:
> Hello, I'm one of the aur maintainers of [balena-etcher](https://aur.archlinux.org/packages/balena-etcher). Currently, I'm planning to build it from source on archlinux following the arch way.
> 
> [Here](https://gist.github.com/sukanka/792fc4b100294ff7a1317e6c04c87302#file-pkgbuild) is my PKGBUILD. But I failed to build this app (more specifically, the module `/build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io`), and the logs are shown below
> 
> update: 1.18.6 can't be built from source either, the error log is the same.
> 
> ```
> npm WARN EBADENGINE Unsupported engine {
> npm WARN EBADENGINE   package: 'balena-etcher@1.10.1',
> npm WARN EBADENGINE   required: { node: '>=14 < 16' },
> npm WARN EBADENGINE   current: { node: 'v18.12.1', npm: '8.19.2' }
> npm WARN EBADENGINE }
> npm WARN EBADENGINE Unsupported engine {
> npm WARN EBADENGINE   package: 'etcher-sdk@7.4.0',
> npm WARN EBADENGINE   required: { node: '>=14 < 18' },
> npm WARN EBADENGINE   current: { node: 'v18.12.1', npm: '8.19.2' }
> npm WARN EBADENGINE }
> npm WARN EBADENGINE Unsupported engine {
> npm WARN EBADENGINE   package: '@balena/node-beaglebone-usbboot@2.0.1',
> npm WARN EBADENGINE   required: { node: '>=14 < 18' },
> npm WARN EBADENGINE   current: { node: 'v18.12.1', npm: '8.19.2' }
> npm WARN EBADENGINE }
> npm WARN deprecated querystring@0.2.0: The querystring API is considered Legacy. new code should use the URLSearchParams API instead.
> npm WARN deprecated @types/puppeteer@7.0.4: This is a stub types definition. puppeteer provides its own type definitions, so you do not need this installed.
> npm WARN deprecated stable@0.1.8: Modern JS already guarantees Array#sort() is a stable sort, so this library is deprecated. See the compatibility table on MDN: https://developer.mozilla.org/en-US/docs/Web/JavaScript/Reference/Global_Objects/Array/sort#browser_compatibility
> npm WARN deprecated raven@2.6.4: Please upgrade to @sentry/node. See the migration guide https://bit.ly/3ybOlo7
> npm WARN deprecated raven-js@3.27.2: Please upgrade to @sentry/browser. See the migration guide https://bit.ly/3ybOlo7
> npm WARN deprecated @braintree/sanitize-url@3.1.0: Potential XSS vulnerability patched in v6.0.0.
> npm WARN deprecated core-js@2.6.12: core-js@<3.23.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Some versions have web compatibility issues. Please, upgrade your dependencies to the actual version of core-js.
> npm WARN deprecated @types/ajv-keywords@3.5.0: This is a stub types definition. ajv-keywords provides its own type definitions, so you do not need this installed.
> npm WARN deprecated @npmcli/move-file@2.0.1: This functionality has been moved to @npmcli/fs
> npm WARN deprecated asar@3.2.0: Please use @electron/asar moving forward.  There is no API change, just a package name change
> npm WARN deprecated svgo@1.3.2: This SVGO version is no longer supported. Upgrade to v2.x.x.
> npm WARN deprecated coffee-script@1.12.7: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
> npm WARN deprecated tslint@6.1.3: TSLint has been deprecated in favor of ESLint. Please see https://github.com/palantir/tslint/issues/4534 for more information.
> npm WARN deprecated electron-rebuild@3.2.9: Please use @electron/rebuild moving forward.  There is no API change, just a package name change
> npm WARN deprecated electron-notarize@1.0.0: Please use @electron/notarize moving forward.  There is no API change, just a package name change
> npm WARN deprecated debug@4.2.0: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
> npm WARN deprecated uuid@3.3.2: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
> npm WARN deprecated core-js@1.2.7: core-js@<3.23.3 is no longer maintained and not recommended for usage due to the number of issues. Because of the V8 engine whims, feature detection in old core-js versions could cause a slowdown up to 100x even if nothing is polyfilled. Some versions have web compatibility issues. Please, upgrade your dependencies to the actual version of core-js.
> npm WARN deprecated coffee-script@1.11.1: CoffeeScript on NPM has moved to "coffeescript" (no hyphen)
> npm WARN deprecated uuid@3.4.0: Please upgrade  to version 7 or higher.  Older versions may use Math.random() in certain circumstances, which is known to be problematic.  See https://v8.dev/blog/math-random for details.
> npm WARN deprecated debug@3.2.6: Debug versions >=3.2.0 <3.2.7 || >=4 <4.3.1 have a low-severity ReDos regression when used in a Node.js environment. It is recommended you upgrade to 3.2.7 or 4.3.1. (https://github.com/visionmedia/debug/issues/797)
> npm notice 
> npm notice New major version of npm available! 8.19.2 -> 9.1.2
> npm notice Changelog: <https://github.com/npm/cli/releases/tag/v9.1.2>
> npm notice Run `npm install -g npm@9.1.2` to update!
> npm notice 
> npm ERR! code 1
> npm ERR! path /build/balena-etcher/src/etcher-1.10.1/node_modules/@ronomon/direct-io
> npm ERR! command failed
> npm ERR! command sh -c -- node-gyp rebuild
[... cut ...]
> �[1m�[31m==> ERROR:�(B�[m�[1m A failure occurred in build().�(B�[m
> �[1m    Aborting...�(B�[m
> ```
> 
>     * **Etcher version:** 1.10.1
> 
>     * **Operating system and architecture:**  archlinux  Kernel: 6.0.9-zen1-1-zen  x86_64
> 
>     * **Image flashed:** None
> 
>     * **What do you think should have happened:**  build the app successfully
> 
>     * **What happened:** The build process failed
> 
>     * **Do you see any meaningful error information in the DevTools?** No

unset MAKEFLAGS may fill fix this problem.

sukanka, 2023-04-07:
> unset MAKEFLAGS may fix this problem.

Thanks! This solves the problem. I'm now able to build it from the source!

closed 2023-04-07 by sukanka