balena-io/etcher #3884: AppImage package for version 1.10.1 initialises with the main program window blank.

H-andy, 2022-11-22:
- Etcher version: 1.10.1

- Operating system and architecture:

  OS: Linux Mint 20.3 x64
  Kernel: 5.15.0-53-generic
  Shell: bash 5.0.17
  Desktop: Cinnamon 5.2.7
  WM: Muffin
  CPU: Intel Core i7-6700K
  GPU: NVIDIA GTX 1080
  RAM: 64251MiB

- Image flashed:

  None

- What do you think should have happened:

  Appimage should have initialised correctly.

- What happened:

  - Downloaded the appimage:

    $ wget -c -N -q --show-progress https://github.com/balena-io/etcher/releases/download/v1.10.1/balenaEtcher-1.10.1-x64.AppImage

  - Enabled it for execution:

    $ chmod 744 ./balenaEtcher-1.10.1-x64.AppImage

  - Executed appimage from the terminal:

    $ ./balenaEtcher-1.10.1-x64.AppImage

  - Terminal output:

    ready-to-show: 772.65ms
  
    The resulting program window has a window title of balenaEtcher with minimise and close buttons,
    but the body of the window is blank/white with no details or objects, i.e.

    [image]

  - For comparison, performing the same actions for appimage version 1.10.0:

    Terminal output:

      ready-to-show: 954.511ms

      The program window is displayed as normal, i.e.

      [image]

  - Repeating the execution of appimage versions 1.10.1 and 1.10.0 from the terminal, the ready-to-show output
  for 1.10.0 was always around 220ms-240ms greater than that for version 1.10.1.

- Do you see any meaningful error information in the DevTools?

  Console:

    index.css:1 Failed to load resource: net::ERR_FILE_NOT_FOUND         _<<< this error is also seen in version 1.10.0>>>_
    electron/js2c/asar_bundle.js:5 Uncaught Error: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required
    by /tmp/.mount_balenaAkbBw9/resources/app/generated/modules/@ronomon/direct-io/binding.node)
      at process.func [as dlopen] (electron/js2c/asar_bundle.js:5)
      at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138)
      at Object.func [as .node] (electron/js2c/asar_bundle.js:5)
      at Module.load (internal/modules/cjs/loader.js:935)
      at Module._load (internal/modules/cjs/loader.js:776)
      at Function.f._load (electron/js2c/asar_bundle.js:5)
      at Function.o._load (electron/js2c/renderer_init.js:33)
      at Module.require (internal/modules/cjs/loader.js:959)
      at require (internal/modules/cjs/helpers.js:88)
      at Object.<anonymous> (gui.js:1)

Etaash-mathamsetty, 2022-11-22:
You have to update your distro to one that ships glibc > 2.33

jadupl2, 2022-11-25:
Still same problem under 1.10.2
I am running Ubuntu 20.04 with latest update.
So I have to go back to 1.8.14 that work under Ubuntu 20.04

Etaash-mathamsetty, 2022-11-26:
> Still same problem under 1.10.2 I am running Ubuntu 20.04 with latest update. So I have to go back to 1.8.14 that work under Ubuntu 20.04

like I said, you need to update to a newer version of Ubuntu (ex: 22.04 or 22.10), which ships glibc > 2.33

theofficialgman, 2022-11-27:
> > Still same problem under 1.10.2 I am running Ubuntu 20.04 with latest update. So I have to go back to 1.8.14 that work under Ubuntu 20.04
> 
> like I said, you need to update to a newer version of Ubuntu (ex: 22.04 or 22.10), which ships glibc > 2.33

no, this is a packaging issue. there is quite literally 0 reason that the github actions are running on 22.04 (which they are now since github switched ubuntu-latest to mean ubuntu-22.04 instead of ubuntu-20.04 in the past few weeks).

this needs changing https://github.com/balena-io/etcher/blob/76d22280dc6f67b36b30c22bcc3a75139fa039e2/.github/workflows/flowzone.yml#L16 it really should be switched to ubuntu-18.04 for as long as it exists

Etaash-mathamsetty, 2022-11-27:
> no, this is a packaging issue. there is quite literally 0 reason that the github actions are running on 22.04 (which they are now since github switched ubuntu-latest to mean ubuntu-22.04 instead of [...]

It's not a packaging issue (you can't package glibc), and your distro is far too out of date because of how old the glibc is, please update it

theofficialgman, 2022-11-27:
> It's not a packaging issue (balena can't package glibc), and your distro is far too out of date because of how old the glibc is, please update it It's not balena's fault that the glibc version requirements have increased most distro have been shipping it for multiple years at this point

please refrain from commenting without doing the research. I am a distro packager. this is a packaging issue. the glibc version requirement is inherited based on the glibc version installed on the buildserver. in this case, github actions ubuntu 22.04 is utilzed so the compiler ends up using glibc 2.33 functions. if the buildserver was switched to ubuntu 18.04 (a simple one line change and everything else would continue to function), the mimimum glibc version would be reduced to 2.27.

this is a balena-ci issue

Etaash-mathamsetty, 2022-11-27:
> > It's not a packaging issue (balena can't package glibc), and your distro is far too out of date because of how old the glibc is, please update it It's not balena's fault that the glibc version requirements have increased most distro have been shipping it for multiple years at this point
> 
> please refrain from commenting without doing background research on how software is compiled. You "I use arch btw" users are no help.
> 
> I am a distro packager. this is a packaging issue. the glibc version requirement is inherited based on the glibc version installed on the buildserver. in this case, github actions ubuntu 22.04 is utilzed so the compiler ends up using glibc 2.33 functions. if the buildserver was switched to ubuntu 18.04 (a simple one line change and everything else would continue to function), the mimimum glibc version would be reduced to 2.27.
> 
> this is a balena-ci issue

Oh you phrased that wrong, I thought something else about balena had changed, well in that case you are correct
Sorry for misunderstanding

JPvRiel, 2022-11-27:
Related to this, is there no way to stop the auto-update feature the goes ahead and breaks things?

E.g. v1.7.9 still linked at <https://www.balena.io/etcher/> runs fine on Ubuntu 20.04 LTS, and doesn't need a very recent libc, but somehow, it defaults to updating and replacing the AppImage with v 1.10.2. And it auto deletes the older working version AppImage after updating. It also seems the CLI options have been deprecated/removed, so there's no way to turn off this auto-update behavior? This is actually quite shocking. Users should have an option to auto-update. It's also a bit of violation to force updates on users without consent.

aethernet, 2022-12-01:
JPvRiel, auto-update is on by default, but it's dead simple to turn in off with a toggle in settings.

aethernet, 2022-12-01:
It would makes total sense to add a flag to turn it off on boot tho.

aethernet, 2022-12-01:
thanks @theofficialgman for suggesting the fix, i've made the PR here : https://github.com/balena-io/etcher/pull/3893

Etaash-mathamsetty, 2022-12-01:
whoops I forgot about this thread, thanks for helping out @aethernet

JPvRiel, 2022-12-01:
> it's dead simple to turn in off with a toggle in settings.

Indeed, I figured that out after about the 3rd SBC flash where I was annoyed by having to re-download an older version again after it was updated/replaced it automatically. Upon first use, it's already gone and downloaded the new the version without any prompt and after closing it's by then deleted and replaced the AppImage with the new version (which in my case had regressions/wasn't working due to gcc...)

Auto-update is a nice feature, just that, before using and doing things to a system, prompting the user is the etiquette I've come to expect from programs. E.g. "Would you like to install the new update v n.n.n: Y/N". Just a suggestion.

tehkillerbee, 2022-12-02:
I had the same issue when re-running 1.10.1 on Ubuntu 20.04. I was surprised, since I knew this version was working previously.   I assume that it has been updated to the latest version that requires >glibc 2.33, without any prompt.

```
Failed to load resource: net::ERR_FILE_NOT_FOUND
electron/js2c/asar_bundle.js:5 Uncaught Error: /lib/x86_64-linux-gnu/libc.so.6: version `GLIBC_2.33' not found (required by /tmp/.mount_balenaIgyeNX/resources/app/generated/modules/@ronomon/direct-io/binding.node)
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5)
    at Module.load (internal/modules/cjs/loader.js:935)
    at Module._load (internal/modules/cjs/loader.js:776)
    at Function.f._load (electron/js2c/asar_bundle.js:5)
    at Function.o._load (electron/js2c/renderer_init.js:33)
    at Module.require (internal/modules/cjs/loader.js:959)
    at require (internal/modules/cjs/helpers.js:88)
    at Object.<anonymous> (gui.js:1)
```

aethernet, 2022-12-02:
Hello, 

This has just been fixed in https://github.com/balena-io/etcher/pull/3893

Sorry for the inconvenience.

closed 2022-12-02 by aethernet