balena-io/etcher #3885: App doesn't install on Linux(Ubuntu 22.04)

saki-osive, 2022-11-24:
- **Etcher version:** : v1.10.1
- **Operating system and architecture:** : x86_64 Ubuntu 22.04 LTS
- **Image flashed:**  : None
- **What do you think should have happened:**  : App doesn't start
- **What happened:** : App doesn't install via AppImage or .deb package installation methods.
- **Do you see any meaningful error information in the DevTools?**  : Nope

gurashish1singh, 2022-11-28:
Can confirm this error. Followed the README steps to add ppa and install the app through the terminal. It fails while assigning permissions:
[image]

prissypants, 2022-12-03:
W: Failed to fetch https://deb.etcher.io/dists/stable/InRelease  Certificate verification failed: The certificate is NOT trusted. The certificate chain uses expired certificate.  Could not handshake: Error in the certificate verification. [IP: 52.41.41.147 443]
W: Some index files failed to download. They have been ignored, or old ones used instead.
merrill@merrill-HP-Pavilion-Laptop-17-ar0xx:~$ sudo apt install balena-etcher-electron
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package balena-etcher-electron

lurch, 2022-12-04:
`deb.etcher.io` was the old BinTray repository, not the new CloudSmith repository? Perhaps one of the Etcher developers ought to update #3501 with instructions for disabling the old BinTray repository (if the user had previously enabled it) ?

prissypants, 2022-12-04:
Finally got it up after going round and round it both on my laptop and pc. No problems before with etcher until I changed both systems to 22.04

Thx for reply

-------- Original Message --------

ja7833, 2022-12-15:
balenaEtcher also does not install on Zorin 16.2/Ubuntu 20.04.3

I downloaded the .AppImage, enabled the “Allow executing file as program” option in permissions, and it just opens an empty window.

See also https://forums.balena.io/t/balenaetcher-not-working-on-zorin-os/365919/5

lurch, 2022-12-15:
See https://github.com/balena-io/etcher/issues/3884#issuecomment-1335324287 ?

aethernet, 2022-12-21:
This is fixed in 1.10.6 (just manually tested the 3 latest Ubuntu LTS 18.04, 20.04 and 22.04 to be sure)

closed 2022-12-21 by aethernet

lurch, 2022-12-22:
Just out of curiosity, do versions of Etcher after #3935 still run on Ubuntu 18.04 LTS?