balena-io/etcher #3888: No way to disable auto-update

JPvRiel, 2022-11-27:
- Version: 1.7.9
- OS: Ubuntu 20.04.5 LTS
- Expected: Should have been prompted for consent to update
- What happens: On first time AppImage launch, auto-update processes without any consent from the user

This relates to #3884 where a known working good version of the app is force auto-upgraded to a version known to have issues.

rradar, 2022-12-01:
Not only that.. but also **not possible to avoid data leakage** to the cloud (other peoples computer) with this program.

Luckily lot's of viable alternatives are out there which actually mostly outperform etcher in terms of privacy, performance, size, etc.

aethernet, 2022-12-01:
Auto update is trivial to disable with a toggle in the settings.

closed 2022-12-01 by aethernet

JPvRiel, 2022-12-01:
I admittedly titled the issue poorly. There is a way to disable updating. My concern is that it's not possible to disable it on first use or via the CLI, e.g. it will always update at least once when working from a fresh first time install. Not a major bug, just not good etiquette to not at least prompt upon first use and inform the user your going to be changing stuff.