balena-io/etcher #3889: End to end tests (incl. flashing a real USB device)

ab77, 2022-11-27:
Currently the Spectron framework (Selenium/WebDriver for Electron) only performs a basic test window visibility test:
https://github.com/balena-io/etcher/blob/master/tests/spectron/runner.spec.ts

This test adds very little confidence as to operability/functionality of a release. The framework is also [deprecated](https://www.electronjs.org/blog/spectron-deprecation-notice).

We need to implement a reasonable e2e test for Etcher using a [supported](https://www.electronjs.org/pt/docs/latest/tutorial/using-selenium-and-webdriver) framework, which tests an actual user journey through the product (i.e. flashing a USB drive with an image). We can instrument a self-hosted GitHub actions runner for this with access to physical USB device(s).

lurch, 2022-11-27:
I guess https://github.com/balena-io/autohat-board could also be used to automatically simulate inserting / removing a USB device from a computer? :thinking:

mcraa, 2023-01-02:
https://github.com/balena-io-experimental/etcher-experimental/blob/middlewares/playwright.config.ts