balena-io/etcher #3892: Etcher 1.10.2 is broken

willfrazer, 2022-12-01:
- **1.10.2:** 
- **Windows 10 64bit:** 
- **USB bootable image:** 
- **Flashing a 2GB bootable USB image that contains a ~400MB CPIO archive containing scripts & tar.gz files. After the image is produced , the USB stick does boot but the CPIO archive md5sum is wrong and files inside it are corrupted. When testing the same image with the RUFUS tool, raspberry PI imager tool OR just DD, the md5sum is consistently correct and works as expected. We repeated the test multiple times (different media, USB adaptors, host systems etc) and etcher was the sole reason why this occurred** 
- **Image contained a corrupted file that was unusable:**

rradar, 2022-12-01:
So why not stick with the privacy friendly (and often more performant) programs you named @willfrazer?

willfrazer, 2022-12-05:
@aethernet Unfortunately, it is proprietary. The description above is the closest that defines what the image contains. What I can say is that we experienced the issue in 3 different geographies on completely different systems. The md5sum of the file was always inconsistent with etcher but reliable with the other tools mentioned above. We have recommended etcher for some time as it is simple enough for non-Linux & non-technical people to use, that's the main reason we would like it fixed - and it has worked reliably in the past. This latest version lost me several days work assuming the problem was in something that I did. I only realised it was etcher when I used the most recent version

willfrazer, 2022-12-06:
@aethernet  I meant to say that should a proposed resolution to the issue be identified, I am happy to test any interim builds against the image that has been previously failing for me.

net147, 2022-12-17:
You can also try v1.12.1 or later from https://github.com/balena-io/etcher/releases and disable the setting "Trim ext{2,3,4} partitions before writing (raw images only)".

willfrazer, 2022-12-20:
@net147 https://github.com/balena-io/etcher/issues/3942#issuecomment-1348601617 worked OK. We will test v1.12.1 shortly. Thanks for your suggestions.

aethernet, 2022-12-26:
Thanks, @net147. 
I was going to propose something similar.

closed 2023-07-12 by dfunckt