balena-io/etcher #3900: 'Error Opening Source' with Version 1.10.6

brysonboettger, 2022-12-05:
- Etcher version: 1.10.6 
- Windows 11 on an ASUS VivoBook -- see image for computer specs
- .bz2 Firmware file

When trying to "select image from file," it gave me the error: 'Error Opening Source:... Error: Cannot read property '0' of undefined.' I reverted back to V1.7.0 (After reading a [forum](https://forums.balena.io/t/flash-failed-cannot-read-property/350420) from december '21) and the error did not occur. I was able to flash my image as desired after reverting back. 

If this could be fixed on the latest Balena Version that would be great. 

Please see the attached screenshots for the computer specs/error information.
[image]
[image]

Seigiard, 2022-12-13:
have same issue on macos with versions 1.10.x and  v1.11.10

v. 1.7.0 works well

lurch, 2022-12-13:
Looks to be related to #3901 which is also using a `.bz2` file :thinking:   Does it work if you manually decompress the image before trying to flash it with Etcher?

visualnaut, 2022-12-19:
> Looks to be related to #3901 which is also using a `.bz2` file 🤔 Does it work if you manually decompress the image before trying to flash it with Etcher?

Happened also to me. And yes I think choosing `.bz2` file is the issue. If you decompress the file first, it worked.

mcraa, 2023-01-02:
same as https://github.com/balena-io/etcher/issues/3901#issuecomment-1369173626

closed 2023-01-02 by mcraa