balena-io/etcher #3901: Error: Cannot read property '0' of undefined on macOS

edwardthehuman, 2022-12-05:
- **Etcher version:** 1.10.6
- **Operating system and architecture:** macOS 10.15.7
- **Image flashed:** [steamdeck-recovery-4.img.bz2](https://store.steampowered.com/steamos/download/?ver=steamdeck&snr=)
- **What do you think should have happened:**  file should've opened when clicking 'Flash from file'
- **What happened:** Error: Cannot read property '0' of undefined
- **Do you see any meaningful error information in the DevTools?** 
 

> Failed to load resource: net::ERR_NAME_NOT_RESOLVED
> api.balena-cloud.com/mixpanel/track/?data=eyJldmVudCI6ICJPcGVuIGltYWdlIHNlbGVjdG9yIiwicHJvcGVydGllcyI6IHsiJG9zIjogIk1hYyBPUyBYIiwiJGJyb3dzZXIiOiAiQ2hyb21lIiwiJGN1cnJlbnRfdXJsIjogImZpbGU6Ly8vQXBwbGljYXRpb25zL2JhbGVuYUV0Y2hlci5hcHAvQ29udGVudHMvUmVzb3VyY2VzL2FwcC9nZW5lcmF0ZWQvaW5kZXguaHRtbCIsIiRicm93c2VyX3ZlcnNpb24iOiA4OSwiJHNjcmVlbl9oZWlnaHQiOiA4MDAsIiRzY3JlZW5fd2lkdGgiOiAxMjgwLCJtcF9saWIiOiAid2ViIiwiJGxpYl92ZXJzaW9uIjogIjIuMzAuMSIsInRpbWUiOiAxNjcwMjY2NDYzLjIzNywiZGlzdGluY3RfaWQiOiAiMTg0ZTNhM2ExODkyMDUtMGEyODliNzQxYWM3ODktNjE5Nzg0ZC1mYTAwMC0xODRlM2EzYTE4YTE4MCIsIiRkZXZpY2VfaWQiOiAiMTg0ZTNhM2ExODkyMDUtMGEyODliNzQxYWM3ODktNjE5Nzg0ZC1mYTAwMC0xODRlM2EzYTE4YTE4MCIsIiRpbml0aWFsX3JlZmVycmVyIjogIiRkaXJlY3QiLCIkaW5pdGlhbF9yZWZlcnJpbmdfZG9tYWluIjogIiRkaXJlY3QiLCJWZXJzaW9uIjogIjEuMTAuNiIsIkFyY2giOiAieDY0IiwiTm9kZSI6ICJ2MTQuMTYuMCIsIk9zIFBsYXRmb3JtIjogImRhcndpbiIsIk9zIFJlbGVhc2UiOiAiMTkuNi4wIiwiQ3B1IENvcmVzIjogNCwiVG90YWwgTWVtb3J5IjogODU4OTkzNDU5MiwiU3RhcnQgRnJlZSBNZW1vcnkiOiA5ODI0OTExMzYsIkhvc3QgQXJjaCI6ICJ4NjQiLCJMb2NhbGUiOiAiZW5fVVMiLCJFbGVjdHJvbiI6ICIxMi4yLjMiLCJTYW1wbGUiOiAwLjEsIkFwcGxpY2F0aW9uIFNlc3Npb24gVXVpZCI6ICIwNzNlMzA1OS1kNzk4LTRhNDYtOThlYy05YzBlNDJmZWI5ZTQiLCJGbGFzaGluZyBXb3JrZmxvdyBVdWlkIjogIjFmNmYzNzczLTY2YjMtNDM2YS04YWZlLTEwZTcxZWY1ZjY0YiIsInRva2VuIjogImJhbGVuYS1ldGNoZXIifX0%3D&ip=1&_=1670266463239:1 Failed to load resource: net::ERR_NAME_NOT_RESOLVED
> gui.js:46 {"stack":"TypeError: Cannot read property '0' of undefined\n    at f (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:160064)\n    at Object.bzip2.decompress (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:157999)\n    at decompressBlock (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149084)\n    at decompressAndQueue (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149300)\n    at Stream.end (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149786)\n    at _end (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:150900)\n    at Stream.stream.end (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:151062)\n    at BlockReadStream.onend (internal/streams/readable.js:684:10)\n    at Object.onceWrapper (events.js:421:28)\n    at BlockReadStream.emit (events.js:315:20)\n    at StreamLimiter._transform (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:1:36667)\n    at StreamLimiter.Transform._read (internal/streams/transform.js:205:10)\n    at StreamLimiter.Transform._write (internal/streams/transform.js:193:12)\n    at writeOrBuffer (internal/streams/writable.js:358:12)\n    at StreamLimiter.Writable.write (internal/streams/writable.js:303:10)\n    at Stream.ondata (internal/streams/legacy.js:19:31)\n    at Stream.emit (events.js:315:20)\n    at drain (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:150565)\n    at Stream.stream.queue.stream.push (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:150693)\n    at file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149336\n    at decompressBlock (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149181)\n    at decompressAndQueue (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149300)\n    at Stream.write (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:149685)\n    at Stream.stream.write (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:35:150396)\n    at BlockReadStream.ondata (internal/streams/readable.js:719:22)\n    at BlockReadStream.emit (events.js:315:20)\n    at addChunk (internal/streams/readable.js:309:12)\n    at readableAddChunk (internal/streams/readable.js:284:9)\n    at BlockReadStream.Readable.push (internal/streams/readable.js:223:10)\n    at BlockReadStream._read (file:///Applications/balenaEtcher.app/Contents/Resources/app/generated/gui.js:1:28563)","message":"Cannot read property '0' of undefined"}
> console.<computed> @ gui.js:46
> o95242.ingest.sentry.io/api/4504259662970880/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=1d7ea0cf1b174f939caed9e80ceab255:1 Failed to load resource: net::ERR_NAME_NOT_RESOLVED

brysonboettger, 2022-12-06:
I had the same issue on a computer running Windows 11 (see Issue 3900). My main work laptop is a Macbook Pro, and I had an older version installed (V1.7.9) with no issues at all. After seeing this issue referencing the error on macOS, I just downloaded the latest Balena Etcher Version for MacOS and got the same issue as @edwardthehuman was referencing.

edwardthehuman, 2022-12-06:
> I had the same issue on a computer running Windows 11 (see Issue 3900). My main work laptop is a Macbook Pro, and I had an older version installed (V1.7.9) with no issues at all. After seeing this issue referencing the error on macOS, I wanted to test it and just downloaded the latest Balena Etcher Version for MacOS and got the same issue as @edwardthehuman was referencing.

I saw a comment from a forum to try version 1.7.0 so I downloaded it. It worked on that version!

brysonboettger, 2022-12-06:
> I saw a comment from a forum to try version 1.7.0 so I downloaded it. It worked on that version!

I did the exact same thing and it also worked for me. I still think this is an issue they need to address for the current version

zjwbupt, 2022-12-20:
Same error when I try to get my steamdeck img done.
Tried latest version from belena website, same error. Latest version from github, same error.
Download 1.7.0 version from Github release, it worked.
[image]

maxpavlov, 2022-12-22:
Okay, the problem is trully on 1.7.0+, but how did you guys figure out that 1.7.0 works? There is no was someone tried all the versions down to late 2021? Why 1.7.0 was tested and not 1.8.0, for instance? Thanks.

brysonboettger, 2022-12-23:
> Okay, the problem is trully on 1.7.0+, but how did you guys figure out that 1.7.0 works? There is no was someone tried all the versions down to late 2021? Why 1.7.0 was tested and not 1.8.0, for instance? Thanks.

I certainly did not try all of the versions. I simply found [this forum post](https://forums.balena.io/t/flash-failed-cannot-read-property/350420) which mentioned that 1.7.0 worked, so that is why I went all the way back to that version. I would feel as though more recent versions may work, but I didn't want to take the time to try to figure out the latest version I could use that didn't have this issue.

lurch, 2022-12-26:
See also #3900 - looks like the problem might be isolated to `.bz2` files?

13r, 2022-12-27:
> Okay, the problem is trully on 1.7.0+, but how did you guys figure out that 1.7.0 works? There is no was someone tried all the versions down to late 2021? Why 1.7.0 was tested and not 1.8.0, for instance? Thanks.

I had the same problem trying to restore the steam deck on macos.
And i found the problem starts with version 1.7.10+. 
With versions 1.7.9 it works fine.

Probably more people will be having this issue because there is a current bug on steam os factory reset and valve suggests Balena Etcher on his FAQ about recovering the deck.

https://help.steampowered.com/en/faqs/view/1b71-edf2-eb6d-2bb3

mcraa, 2023-01-02:
Worked below 1.7.9, works above 1.13.1
Thanks for the report I hope all steamdecks gonna be fine.

closed 2023-01-02 by mcraa

TDuffinNTU, 2023-03-06:
Uncompressing the .bz2 image fixed it for me, but this is only a workaround and might not be clear to newer users who may wonder why etched appears to suggest a disk image will take take  10+ hrs to flash :)

alfred-kargo, 2023-03-07:
> Uncompressing the .bz2 image fixed it for me, but this is only a workaround and might not be clear to newer users who may wonder why etched appears to suggest a disk image will take take 10+ hrs to flash :)

Totally fixed the issue for flashing steamOS image with the solution above 💯

> Worked below 1.7.9, works above 1.13.1 Thanks for the report I hope all steamdecks gonna be fine.

still having issue in 1.18.4 @mcraa 
there is still ongoing issue where the .bz2 source just breaks the flashing process.

mcraa, 2023-03-07:
i am not a maintainer any more, cc @aethernet 
possibly the sdk dependency changed in the newest version

aethernet, 2023-03-09:
Thanks for the ping @mcraa, I've reproduced and will investigate