balena-io/etcher #3902: USB stick unusable after flashing with Balena

ZippoLag, 2022-12-06:
- **Etcher version:** 1.10.2
- **Operating system and architecture:** Windows 10 Home x64
- **Image flashed:** linuxmint-21-cinnamon-64bit.iso
- **What do you think should have happened:** I wanted to have Linux Mint's ISO flashed into a USB stick, instead now I have an unusable USB stick which can't be formatted.
- **What happened:**
I inserted an 8gb USB stick into the port and ran Balena Etcher, flashing the latest linux mint ISO into it. The process claimed to finish successfully, however the drive stopped being recognized by Windows and asked for a format, now reporting that the size is around 2gb, instead of 8gb, and saying that it can't be formatted due to write protection:
[image]
I tried plugging the stick on a Linux machine and it's not recognized either. Nor can I use it to boot as intended.
Curiously, though, if I start Balena again the program still recognizes the drive as 8gb!
[image]

- **Do you see any meaningful error information in the DevTools?** 
The only error I see in the dev tools is regarding a "disable screensaver" event:
`{"stack":"Error: Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'"}`

This may be related to [3241](https://github.com/balena-io/etcher/issues/3241), but I don't use bitlocker or any encryption software.

Is there any way I may recover this device?

lurch, 2022-12-07:
See https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md#recovering-broken-drives and https://www.balena.io/blog/did-etcher-break-my-usb-sd-card/

RV-Levi, 2022-12-09:
Me too, but flashing bodhi Linux on Windows 8.1.

aethernet, 2022-12-21:
See lurch links.

closed 2022-12-21 by aethernet