balena-io/etcher #3908: raspbian 64 installation problem

karllumpkins, 2022-12-07:
- **Etcher version:** 
- **Operating system and architecture:** 
- **Image flashed:** 
- **What do you think should have happened:** 
- **What happened:**
- **Do you see any meaningful error information in the DevTools?**

karllumpkins, 2022-12-07:
get a not found for sudo apt-get install balena-etcher-electron
pi@raspberrypi:~$ sudo apt-get install balena-etcher-electron
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
E: Unable to locate package balena-etcher-electron
pi@raspberrypi:~$

lurch, 2022-12-07:
See #2870

lurch, 2022-12-07:
...but if you're running Raspberry Pi OS, you could just use Raspberry Pi Imager? https://www.raspberrypi.com/software/

karllumpkins, 2022-12-08:
your choices are limited with pi imager, I need etcher.

lurch, 2022-12-09:
You can still select "Use custom" in RPi Imager, if you want to use your own .img file.

lurch, 2022-12-09:
> is that also for a .iso image ?

Yes, see https://github.com/raspberrypi/rpi-imager/blob/qml/src/imagewriter.cpp#L589

karllumpkins, 2022-12-09:
ok thanks, so etcher is dead?

aethernet, 2022-12-21:
@karllumpkins not dead at all you can install using https://github.com/Botspot/pi-apps
Or build it from source.

The balena-etcher-electron wasn't an official package and has been dropped as far as I know

closed 2022-12-21 by aethernet

lurch, 2022-12-21:
Just out of curiosity (because I'm a curious chap), I did a bit of digging and it looks like pi-apps [only has version v1.10.0](https://github.com/Pi-Apps-Coders/files/releases/tag/large-files) of Etcher, which is quite a bit behind the [current version v1.12.7](https://github.com/balena-io/etcher/releases) ?