balena-io/etcher #3942: Re-add setting to disable trim

net147, 2022-12-12:
- **Etcher version:** v1.10.2
- **Operating system and architecture:** Windows x64
- **Image flashed:** Raw SD image with ext4 partition that has unallocated blocks
- **What do you think should have happened:** The entire image is written as-is to the SD card
- **What happened:** Writing of unallocated blocks in the ext{2,3,4} partitions are skipped
- **Do you see any meaningful error information in the DevTools?** No

Currently the option can only be disabled in config.json by setting autoBlockmapping to false. There was previously a setting with the text "trim ext{2,3,4} partitions before writing (raw images only)" but it was removed. The data contained in the unallocated blocks of the image may be important in some use cases such as verifying the checksum of partitions or images.

aviks1919, 2022-12-13:
i have a 16GB SD card firmware image writing to 32GB SD card. After the firmware flashing, the SD card EXT4 partition has some files corruption. However, if i use Win32DiskImager, i did not see this issue. Do you think the issue i encountered is related to your mentioned "unallocated blocks in the ext4 partition are skipped"?

lurch, 2022-12-13:
Might be related to #3892 ? (which is also about Etcher writing corrupt images, which work fine with other image-flashing tools)

net147, 2022-12-13:
For people having issues with Etcher on Windows writing corrupt images, can you please try the following?
1. Close Etcher
2. Press Win+R to open the run dialog
3. Type the following and press enter: notepad %AppData%\balena-etcher\config.json
4. Change "autoBlockmapping": true, to "autoBlockmapping": false,
5. Save the file
6. Start Etcher
7. Try writing the image again

If you want to enable trim again, do the same as above but change autoBlockmapping back to true.

Thanks.

net147, 2022-12-13:
> i have a 16GB SD card firmware image writing to 32GB SD card. After the firmware flashing, the SD card EXT4 partition has some files corruption. However, if i use Win32DiskImager, i did not see this [...]

It could be related. Try what I suggested in https://github.com/balena-io/etcher/issues/3942#issuecomment-1348601617.

aviks1919, 2022-12-14:
> For people having issues with Etcher on Windows writing corrupt images, can you please try the following?
> 
> 1. Close Etcher
> 2. Press Win+R to open the run dialog
> 3. Type the following and press enter: notepad %AppData%\balena-etcher\config.json
> 4. Change "autoBlockmapping": true, to "autoBlockmapping": false,
> 5. Save the file
> 6. Start Etcher
> 7. Try writing the image again
> 
> If you want to enable trim again, do the same as above but change autoBlockmapping back to true. If you are using Etcher on Linux, config.json may be located at ~/.config/balena-etcher-electron/config.json.
> 
> Thanks.

Yeah, this fixed my issue! May i know about this "autoBlockmapping" setting? How do i know when i need to set to true or false?

net147, 2022-12-14:
> Yeah, this fixed my issue! May i know about this "autoBlockmapping" setting? How do i know when i need to set to true or false?

It is always safe to set it to false. Setting it to true will skip writing blocks that are not used in the filesystem to speed up writing images that have a lot of free space. It is true by default for some reason.

lurch, 2022-12-14:
...although if it's causing file-corruption, perhaps it's got buggy in later versions of Etcher? :shrug:

aviks1919, 2022-12-15:
May i know for my case of the SD card has FAT and EXT4 partition, this issue is it affect the EXT4 partition content only?

net147, 2022-12-15:
> May i know for my case of the SD card has FAT and EXT4 partition, this issue is it affect the EXT4 partition content only?

https://github.com/balena-io/etcher/blob/v1.10.2/lib/gui/modules/child-writer.ts#L115
@param {Boolean} autoBlockmapping - whether to trim ext partitions before writing

The documentation says it only affects ext partitions.

closed 2022-12-17 by net147

Hamstek, 2023-01-08:
I can't use persistence in kali-linux-2022.4-live-everything-amd64 after using Balenaetcher v1.13.1.

lurch, 2023-01-08:
> I can't use persistence in kali-linux-2022.4-live-everything-amd64 after using Balenaetcher v1.13.1.

a) This doesn't seem to have anything to do with the rest of the comments in this issue?

b) See https://github.com/balena-io/etcher/blob/master/FAQ.md#how-can-i-configure-persistent-storage