balena-io/etcher #3949: Could the reliance on a working GPU driver be dropped?

markg85, 2022-12-18:
I just tried to run the latest appimage: balenaEtcher-1.10.6-x64.appimage

Which fails because i'm running in CPU mode (don't ask...)
```
libva error: /usr/lib/dri/radeonsi_drv_video.so init failed
[11848:1218/114548.381126:FATAL:gpu_data_manager_impl_private.cc(415)] GPU process isn't usable. Goodbye.
```

Not having this reliance would be helpful in cases where a GPU driver isn't working. A case that is probably quite common when you need a tool like this imho.

rradar, 2022-12-18:
Why bother with etcher at all when no GPU (driver) is present? It will be buggy as hell and a nightmare to work with...

Just use something which integrates natively (without hundreds of megabytes of dependencies, analytics, ads, etc.) - probably [usbimager](https://bztsrc.gitlab.io/usbimager/) will do the job just right for you (yes the whole program is less than a megabyte with gui).

markg85, 2022-12-18:
Well, in my specific case i already had etcher and used it before. Just a change in gpu and driver caused that side to go broken which now has the side effect that etcher didn't work either (wanted to make a live usb for something else)

Conceptually it's like absolute nonsense that a tool like etcher **requires** working gpu drivers... Note that we're talking about the DRM bits even, those that handle hardware video decoding. Oh well.

Thank you for the pointer to usbimager! I was using `dd` now but kinda find it too dangerously low level (typo and you ruin something you regret). Will give your usbimager a shot!

rradar, 2022-12-19:
I must admit at this point the whole existence of etcher is to question (imho). Like you wrote it essentially a fancy gui for `dd` and that's about it. But why ship a EOL browser base (electron) with it and have opt-out telemetry sending information home to the mothership by default? Why presenting ads (which can not be disabled) and why is that program almost always bigger than the things someone wants to write?

Not only in terms of **security** and **privacy** this is a real nightmare but it is also just a **huge waste** of resources :earth_africa:

markg85, 2022-12-19:
Yeah, i fully agree with you @rradar!

It doesn't resolve the issue per se.

Then again, if etcher effectively falls into the EOL abandonware category then fixing a dead horse - so to say - is a waste of time too.

aethernet, 2022-12-21:
@rradar I'm seriously questioning your motivation. I couldn't find one post where you don't try to undermine the work we do or try to push user to use other software.

I totally gets that you don't like etcher, and you're free to express that point of view.
But be warned that constant bashing is not ok and won't be tolerated indefinitely.

Back to the original question, AFAIK this is a requirement of electron, I'll check if we can turn that requirement off as we don't use the feature.

PunchCakee, 2022-12-28:
> I must admit at this point the whole existence of etcher is to question (imho). Like you wrote it essentially a fancy gui for `dd` and that's about it. But why ship a EOL browser base (electron) [...]

Dont worry, i am working on a similar app but it wont use electron it will use tauri + rust and it will have a simple gui that even a grandma can use. Let this shitty app rot in hell

PunchCakee, 2022-12-28:
> @rradar I'm seriously questioning your motivation. I couldn't find one post where you don't try to undermine the work we do or try to push user to use other software.
> 
> I totally get that you don't like etcher, and you're free to express that point of view. But be warned that constant bashing is not ok and won't be tolerated indefinitely.
> 
> Back to the original question, AFAIK this is a requirement of electron, I'll check if we can turn that requirement off as we don't use the feature.

The issue is, why construct such a badly designed app to begin with? 124 mb for an app thats just copy pasting everything into a usb? Electron is one of the worst things that ever happened to the desktop scene good thing tauri is slowly killing it, Not to even mention how these shitty devs never even bothered to fix the linux gpu problem.

mcraa, 2022-12-28:
Wow, these comments are reaching the pinnacle of open-source.
Since etcher is open-source when blaming the "shitty devs" you are blaming everyone, as anyone could contribute to the project, seem like including yourself.
Funny, how confident some comments are about etcher being a GUI for dd, however, it is not. Clearly, you never checked the code.
When etcher was created tauri (and friends) did not exist. Good luck with the tauri development we also experiment with different solutions.

I am glad many people have so much time that can just come here and drive issues offtopic. Sadly they don't have time to search among the issues to find all the answers for the repeated blame they are putting here. And we don't have the time to sink to your level and argue, so this issue will be closed now.

Sorry for the OP that the issue was derailed.
The GPU problem can be solved with an electron update since it is a problem there.
#3945 PR looks good with updated electron hopefully can be merged soon

closed 2022-12-28 by mcraa