balena-io/etcher #3957: Etcher claims "Flash failed" but it didn't fail.. it succeeded

mmikeww, 2022-12-23:
- **Etcher version:** 1.10.2
- **Operating system and architecture:**  Win 10 x64
- **Image flashed:** Tails 5.8
- **What do you think should have happened:** Etcher tells me flash succeeded
- **What happened:** Etcher tells me flash failed

I got the first "flash failed" message when I tried to overwrite an existing flashed USB stick. I figured maybe thats not allowed. Then I opened a brand new stick and got the same error. The "Flashing" progress bar completes fully, then the "Verifying" progress bar also completes fully. Both seem to reach 100%. And then immediately after the Verification part finishes, it says "Flash failed". 

I was pulling my hair out, trying to reformat and repartition these USB sticks. Finally I came across this post:  
https://forums.balena.io/t/error-message-flash-failed/365401/2

There a balena team member suggests to just try it anyway, and that's what I did, and the USB stick booted fine, despite the "failed" message.

Another person in that thread hints that it could be a Windows issue. But even if so, Etcher should do some clever engineering and try to detect that, and not throw a "failed" message when in fact the flash did succeed.

aethernet, 2022-12-26:
Indeed this is weird. 

The forum thread you pointed out ended up being an issue with the older sd card adapter.

Could it be the case here that you're using a quite old USB stick? What was the model?
I'd like to repro, but couldn't so far.

mmikeww, 2022-12-28:
SanDisk Cruzer Glide 16gb
https://www.amazon.com/SanDisk-Flash-Cruzer-Glide-SDCZ60-016G-B35/dp/B007YX9O9O/

Not sure if these are considered "old" or not. When the first one gave the error, I opened a brand new one that was still sealed in packaging.

kylebakerio, 2023-01-02:
I had this flash failed, just opened logs though and I suspect this error is being thrown and causing a 'wrong' failure message after flash succeeds: 

```
gui.js:46 {"stack":"Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'"}
```

lurch, 2023-01-02:
_Maybe_ that was fixed by #3882 ? :shrug:

mcraa, 2023-01-03:
Nope @lurch, that started it, but it has no effect on the functionality.

kylebakerio, 2023-01-07:
Yeah, when I posted that comment, I suspected it was successful, but it turned out it truly wasn't, so my comment should be taken as a different issue.

cretan-ibex, 2023-01-08:
> 

I have the same error output.
Anyone have any clarity on this issue? Any suggestions on what older version to try and download and flash with?
I have tried both 1.13.1 and 1.12.7 both with the same issue.

cretan-ibex, 2023-01-08:
I have tried both 1.13.1 and 1.12.7 both with the same issue as @kylebakerio shows above.

Any suggestion on another version that would work?

adamtha, 2023-12-13:
v1.18.11

still facing the exact same issue on windows 11 and SanDisk cruzer
the flash + verify ends successfully
the screensaver error seems to mark the process as failed

the flashed OS boots and works as expected

```
gui.js:35 Verifying 1 device, 97% at 18.00 MB/s (total 18.00 MB/s) eta in 2s with 0 failed devices
gui.js:35 Verifying 1 device, 98% at 17.50 MB/s (total 17.50 MB/s) eta in 1s with 0 failed devices
gui.js:35 Verifying 1 device, 100% at 18.30 MB/s (total 18.30 MB/s) eta in 0s with 0 failed devices
gui.js:37 Finish: 1292605440
gui.js:37 Terminating IPC server
gui.js:37 Flash results {results: {…}, cancelled: false, skip: false}cancelled: falseresults: {bytesWritten: 1292605440, devices: {…}, errors: Array(1), sourceMetadata: {…}, averageFlashingSpeed: 5664651.252476028}skip: false[[Prototype]]: Object
gui.js:37 Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'
    at o.invoke (node:electron/js2c/renderer_init:57:526)
(anonymous) @ gui.js:37
logException @ gui.js:37
(anonymous) @ gui.js:354
```

phillipremaker, 2024-01-09:
I have the same issue on two Windows 11 machines, with both USB and PCIe based SD card writers and multiple SD cards.

CrazyWolf13, 2024-01-23:
Same for me here.

staffa, 2024-04-30:
As we have established in #4161 (and I would recommend following up there instead), this is likely unrelated to the enable screensaver issue; the failure should instead be present in the flash results object in the line above if opened. Flash results  > results > errors > 0 to see the failure that the program is actually reporting.