balena-io/etcher #3960: event submission rejected with_reason: ProjectId

harirajanv, 2022-12-26:
- **Etcher version:** 
        Version 1.10.6 (1.10.6)

- **Operating system and architecture:** 
    Mac OS Ventura 13.1 (22C65)

- **Image flashed:** 
Samsung_SSD_980_3B4QFXO7.iso.  this was downloaded directly from Samsung website.

- **What do you think should have happened:** 
Open Etcher UI, then select iso, UBB 128gb ( USB SanDisk 3.2Gen1 Media)
opened developer tools.. 
Hit. Flash.
it should have just flashed the empty drive, but failed with below error only seen in developer tools.
on UI, it just returns back to flash screen again as if no-one click on flash yet.

- **What happened:**
got error after it asked to enter the password for the usb flash drive (not sure why it ask for password though)

- **Do you see any meaningful error information in the DevTools?** 

Request:
Request URL: https://xxxxxx.ingest.sentry.io/api/4504259662970880/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=xxxxxxxxxxxxxxxxx
Request Method: POST
Status Code: 403 
Remote Address: xxxxx:443
Referrer Policy: origin

Response:
{"detail":"event submission rejected with_reason: ProjectId"}

mcraa, 2022-12-26:
Thanks for the feedback. It is strange, we will have a look.
The password is needed because managing the drives (mounting, unmounting) requires admin privileges. 

The sentry error should not make the flashing fail. 
(my guess would be there was an error earlier then the error report send request fails)
@otaviojacobi do you know any change on the sentry side which can lead to something like this?

lurch, 2022-12-26:
I guess #3891 might fix this? :shrug:

harirajanv, 2022-12-26:
Thanks guys.. what does this mean to me.. Is this something you would be fixing and releasing a new version of Etcher or do I need to try something else.
@mcraa  yes I was logged in as standard user when I was running that.. 

FYI.
Next what I did it, I tried to write the same .iso to another usb stick of 4GB , thought process being that 128GB usb disk was too big  , though it should have worked. 
So I formatted a 4GB usb stick as. 'exFAT' format. and then tried to action the same.
This time, it again asked me for the password, after entering it, I see that the same sentry.io https url has status code of 200. and returns back to the same screen of select disk to flash

BUT nothing actually written to disk.

[image]

[image]

mcraa, 2022-12-27:
> > Thanks for the feedback. It is strange, we will have a look. The password is needed because managing the drives (mounting, unmounting) requires admin privileges.
> > The sentry error should not make the flashing fail. (my guess would be there was an error earlier then the error report send request fails) @otaviojacobi do you know any change on the sentry side which can lead to something like this?
> 
> Hey @mcraa all work on Sentry is still WIP in #3891 so I do think this is related to my current changes.

I believe you left out a *not* from there 😄 @otaviojacobi ^
I don't think it is related, just since you are more familiar around the sentry dashboard I was curious if you've seen any change there, like project id, notification about API outage, whatever.

harirajanv, 2022-12-28:
Hi @mcraa , @otaviojacobi , @lurch 
Thanks guys for all the replies..  new findings..  This time I logged into an account that was set as admin and tried the same .
as expected of it, it did ask for the password once, so I entered it and  great.. it went through fine and did flash the usb drive.

So I reckon the issue was that previously I used a non-admin account to run the command, which then ask for password, but  fails..  maybe  the password is required at multiple time within the flashing process and it doesn't have the option to prompt for password every time..  so maybe since I used admin the second time, it worked..  Something that might help you  think of the process flow within the flashing flow.

I have another issue, is that even though it flashed the usb stick, it was not bootable , maybe not be related to this Etcher app as such...

Thanks

mcraa, 2022-12-28:
Mac OS is constantly changing and getting more restrictive.
Thanks for your investigation and feedback, probably the rights elevation failed and the request error has nothing to do with the flash starting/not starting.

Can you send a link to the same iso so we can have a look?

harirajanv, 2022-12-31:
> Mac OS is constantly changing and getting more restrictive. Thanks for your investigation and feedback, probably the rights elevation failed and the request error has nothing to do with the flash starting/not starting.
> 
> Can you send a link to the same iso so we can have a look?

you can try this with macOS Ventura 13.1 latest version
https://download.semiconductor.samsung.com/resources/software-resources/Samsung_SSD_980_3B4QFXO7.iso

mcraa, 2023-01-02:
[image]
I was able to boot it, you may need to enable legacy boot in bios.

harirajanv, 2023-01-15:
> [image]
> I was able to boot it, you may need to enable legacy boot in bios.

Did you boot in MAC. ?  I have iMac that didn't work.
or you meant windows.

mcraa, 2023-01-16:
Well macs have UEFI so my bios comment is irrelevant.
I just wanted to show that the drive made from the image is bootable.
Added the extra info because this PC on the picture is set to legacy boot in bios, because sometimes I need that, it does not mean it is needed for this image in particular.

harirajanv, 2023-01-16:
Thanks @mcraa .  I unfortunately don't have windows.  mine is  Mac , and that image was not bootable.
I think etcher software as such is good. just that it doesn't create bootable Mac with that image.
maybe it is Samsung issue that they didn't provide a proper bootable iso image.