balena-io/etcher #3961: AUR package took more than 2 hours to install!

samiurprapon, 2022-12-27:
- **Etcher version:** 1.11.5
- **Operating system and architecture:**  EndeavourOs
- **Image flashed:** No
- **What do you think should have happened:**   `yay -Sy balena-etcher` and Installation done within 2-15 minutes 
- **What happened:** It took more than 143 minutes to install on a machine (Spec. Ryzen 9)
- **Do you see any meaningful error information in the DevTools?**

aethernet, 2022-12-30:
@samiurprapon I'm not familiar with EndeavourOs at all, does it get the source and build everything locally? 
Even in that case it's very long.

aethernet, 2022-12-30:
Reading a little about it, AUR is a community package repository and we're not publishing ourselves. 
I do not know how it's managed or how the build/install process works in that case.

Also, 1.11.5 is a pre-release and is not intended to be distributed to package managers.

I would advise you to get [1.10.6 (the latest stable release)](https://github.com/balena-io/etcher/releases/tag/v1.10.6)

Or to compile from sources.

samiurprapon, 2023-01-02:
@aethernet  EndeavourOS is arch based distro, unlike Manjaro EOS usages mostly arch packages. 

I checked the [PKGBUILD](https://aur.archlinux.org/cgit/aur.git/tree/PKGBUILD?h=balena-etcher&id=21be8d1e9cee9bbfe0d7918171f85e9f0085bd5e) and re-run `yay -S balena-etcher`  to verify which packages it builds from source.  

- `Python-2.7.18` alone took 30 minutes to build. 
-  Electron 13 requires `chromium-91-patchset-5` to build again!  🐸   _chromium itself _1.13 GiB in size_ 

basically building `python` and `electron` took more than 2 hours! I don't think it will do any better on Arch Linux or Manjaro.

samiurprapon, 2023-10-18:
> After about 30min yay is still "syncing projects". Seam to be some sort of source file-syncing. I've gone from 8% to 15%+ storage on a 234GB drive.
> 
> I hope I am able to clear some of these files later...

All I can say rather than using etcher  `AUR` package, AppImage is more convenient.

OskyEdz, 2023-10-18:
> > After about 30min yay is still "syncing projects". Seam to be some sort of source file-syncing. I've gone from 8% to 15%+ storage on a 234GB drive.
> > I hope I am able to clear some of these files later...
> 
> All I can say rather than using etcher `AUR` package, AppImage is more convenient.

I didn't really consider other options as I've had etcher-bin installed since before. But it was recently marked as out-of-date.
That package was a lot faster to install.

irfanhakim-as, 2023-11-10:
I'd be very happy if Balena decides to support/distribute Flatpaks at some point in the future so more people could install and use Etcher on Linux, regardless of distro (even Steam Decks). Other similar utilities on Linux have been a hit-or-miss for me.