balena-io/etcher #3962: cannot install on kubuntu 20.04

safaATcurrus, 2022-12-27:
- **Etcher version:** 
I couldnt even install. I guess the last version
- **Operating system and architecture:**  
linux kubuntu 20.04 amd64
- **Image flashed:** 
I couldnt even install... 
- **What do you think should have happened:** 
I was expecting it to be installed.

- **What happened:**

when I did 

sudo apt-get install balena-etcher-electron

I faced with :

 
chmod: cannot access '/opt/balena-etcher-electron/chrome-sandbox': No such file or directory

Error in file "/usr/share/applications/org.kde.kdeconnect_open.desktop": "*/*" is an invalid MIME type ("*" is an unregistered media type)

I even couldnt install. 

P.S.
I did "'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
>    | sudo -E bash" before this and all seemed to be ok.

One important question: how to run after install? Where is the command written?

My main problem is that I couldnt install it on kubuntu.

mcraa, 2022-12-28:
hi,
sorry for the inconvinience apt still gets the 1.7.9 version which does not use the file what fixes this
https://github.com/balena-io/etcher/blob/master/after-install.tpl
there you see the difference in the path `balenaEtcher` instead of `balena-etcher-electron`

lurch, 2022-12-28:
Duplicate of #3771 ?

aethernet, 2022-12-30:
Hello, 

A quick workaround would be to get the .deb package for the latest stable version on github ([1.10.6](https://github.com/balena-io/etcher/releases/tag/v1.10.6) and install it with `dpkg -i  balena-etcher_1.10.6_amd64.deb .deb`

closed 2023-07-12 by dfunckt