balena-io/etcher #3966: Flash Failed on macOS Ventura 13.0.1

shimoshida, 2023-01-05:
- **Etcher version:** 1.13.1
- **Operating system and architecture:** macOS Ventura 13.0.1 and Intel 
- **Image flashed:**  ubuntu-22.04.1-desktop-amd64.iso
- **What do you think should have happened:** 
Write *.iso to usb successfully.

- **What happened:**

Flash failed by showing nothing(just return start GUI.)
This happens by using other ubuntu versions such as `20.04` and `18.04`.

- **Do you see any meaningful error information in the DevTools?** 

The following logs are all in console
<details>

```
 _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.13.1, Type = dmg
gui.js:37 Thu Jan 05 2023 11:48:21 GMT+0900 (日本標準時) Application start ({"packageType":"dmg","version":"1.13.1","sample":0.1,"applicationSessionUuid":"0be46af1-bf0a-4615-ae82-d6a9095b2bce","flashingWorkflowUuid":"6662fc3b-f3b7-4704-a7f9-b28cc67ddad6"})
gui.js:37 Thu Jan 05 2023 11:49:25 GMT+0900 (日本標準時) Open image selector ({"sample":0.1,"applicationSessionUuid":"0be46af1-bf0a-4615-ae82-d6a9095b2bce","flashingWorkflowUuid":"6662fc3b-f3b7-4704-a7f9-b28cc67ddad6"})
gui.js:37 Thu Jan 05 2023 11:49:34 GMT+0900 (日本標準時) Select image ({"image":{"size":3826831360,"name":"ubuntu-22.04.1-desktop-amd64.iso","hasMBR":true,"partitions":[{"offset":32768,"size":3822108672,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"9240A165-D190-4AB6-8A11-46DC207B42EE","name":"ISO9660","index":1},{"offset":3822141440,"size":4349952,"type":"C12A7328-F81F-11D2-BA4B-00A0C93EC93B","guid":"9240A165-D190-4AB6-8A12-46DC207B42EE","name":"Appended2","index":2},{"offset":3826491392,"size":307200,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"9240A165-D190-4AB6-8A13-46DC207B42EE","name":"Gap1","index":3}],"extension":"iso","path":"/Users/shimodamasayuki/Downloads/ubuntu-22.04.1-desktop-amd64.iso","logo":false,"blockMap":false},"sample":0.1,"applicationSessionUuid":"0be46af1-bf0a-4615-ae82-d6a9095b2bce","flashingWorkflowUuid":"6662fc3b-f3b7-4704-a7f9-b28cc67ddad6"})
gui.js:37 Thu Jan 05 2023 11:49:38 GMT+0900 (日本標準時) Flash ({"image":{"size":3826831360,"name":"ubuntu-22.04.1-desktop-amd64.iso","hasMBR":true,"partitions":[{"offset":32768,"size":3822108672,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"9240A165-D190-4AB6-8A11-46DC207B42EE","name":"ISO9660","index":1},{"offset":3822141440,"size":4349952,"type":"C12A7328-F81F-11D2-BA4B-00A0C93EC93B","guid":"9240A165-D190-4AB6-8A12-46DC207B42EE","name":"Appended2","index":2},{"offset":3826491392,"size":307200,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"9240A165-D190-4AB6-8A13-46DC207B42EE","name":"Gap1","index":3}],"extension":"iso","path":"/Users/shimodamasayuki/Downloads/ubuntu-22.04.1-desktop-amd64.iso"},"drives":[{"size":15518924800,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[{"path":"/Volumes/NO NAME","label":"NO NAME"}],"description":"Generic Flash Disk Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"62b36f00-0112-4252-97c5-04d9c6b22b3d","status":"started","flashInstanceUuid":"62b36f00-0112-4252-97c5-04d9c6b22b3d","sample":0.1,"applicationSessionUuid":"0be46af1-bf0a-4615-ae82-d6a9095b2bce","flashingWorkflowUuid":"6662fc3b-f3b7-4704-a7f9-b28cc67ddad6"})
gui.js:37 Elevating command: /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper (Renderer).app/Contents/MacOS/balenaEtcher Helper (Renderer) /Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js
gui.js:37 {"stack":"Error: Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'"}
console.<computed> @ gui.js:37
logException @ gui.js:37
(anonymous) @ gui.js:354
gui.js:37 POST https://o95242.ingest.sentry.io/api/4504436020019200/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=a89e60810d444088ae7957d3a84837ee 403
(anonymous) @ gui.js:37
_makeRequest @ gui.js:37
_sendProcessedPayload @ gui.js:37
_send @ gui.js:37
_processException @ gui.js:37
_handleStackInfo @ gui.js:37
captureException @ gui.js:37
_promiseRejectionHandler @ gui.js:37
gui.js:37 POST https://o95242.ingest.sentry.io/api/4504436020019200/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=a89e60810d444088ae7957d3a84837ee 403
(anonymous) @ gui.js:37
_makeRequest @ gui.js:37
_sendProcessedPayload @ gui.js:37
_send @ gui.js:37
_processException @ gui.js:37
_handleStackInfo @ gui.js:37
captureException @ gui.js:37
captureException @ gui.js:37
logException @ gui.js:37
(anonymous) @ gui.js:354
gui.js:35  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:37 Thu Jan 05 2023 11:49:40 GMT+0900 (日本標準時) TypeError: Converting circular structure to JSON
    --> starting at object with constructor 'Object'
    --- property 'top' closes the circle
gui.js:37 Uncaught RangeError: Maximum call stack size exceeded
    at gui.js:37
    at Array.forEach (<anonymous>)
    at step (gui.js:37)
    at gui.js:37
    at Array.forEach (<anonymous>)
    at step (gui.js:37)
    at gui.js:37
    at Array.forEach (<anonymous>)
[... 616 lines cut ...]
(anonymous) @ gui.js:37
notifyHandlers @ gui.js:37
traceKitWindowOnError @ gui.js:37
gui.js:37 Terminating IPC server
gui.js:37 Flash results {cancelled: true, skip: false}
gui.js:37 Thu Jan 05 2023 11:49:41 GMT+0900 (日本標準時) Elevation cancelled ({"image":{"size":3826831360,"name":"ubuntu-22.04.1-desktop-amd64.iso","hasMBR":true,"partitions":[{"offset":32768,"size":3822108672,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"9240A165-D190-4AB6-8A11-46DC207B42EE","name":"ISO9660","index":1},{"offset":3822141440,"size":4349952,"type":"C12A7328-F81F-11D2-BA4B-00A0C93EC93B","guid":"9240A165-D190-4AB6-8A12-46DC207B42EE","name":"Appended2","index":2},{"offset":3826491392,"size":307200,"type":"EBD0A0A2-B9E5-4433-87C0-68B6B72699C7","guid":"9240A165-D190-4AB6-8A13-46DC207B42EE","name":"Gap1","index":3}],"extension":"iso","path":"/Users/shimodamasayuki/Downloads/ubuntu-22.04.1-desktop-amd64.iso"},"drives":[{"size":15518924800,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk2","error":null,"isReadOnly":false,"displayName":"/dev/disk2","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk2","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[{"path":"/Volumes/NO NAME","label":"NO NAME"}],"description":"Generic Flash Disk Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"62b36f00-0112-4252-97c5-04d9c6b22b3d","status":"cancel","flashInstanceUuid":"62b36f00-0112-4252-97c5-04d9c6b22b3d","sample":0.1,"applicationSessionUuid":"0be46af1-bf0a-4615-ae82-d6a9095b2bce","flashingWorkflowUuid":"6662fc3b-f3b7-4704-a7f9-b28cc67ddad6"})
gui.js:37 {"stack":"Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'"}
console.<computed> @ gui.js:37
logException @ gui.js:37
(anonymous) @ gui.js:354
gui.js:37 POST https://o95242.ingest.sentry.io/api/4504436020019200/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=a89e60810d444088ae7957d3a84837ee 403
(anonymous) @ gui.js:37
_makeRequest @ gui.js:37
_sendProcessedPayload @ gui.js:37
_send @ gui.js:37
_processException @ gui.js:37
_handleStackInfo @ gui.js:37
captureException @ gui.js:37
captureException @ gui.js:37
logException @ gui.js:37
(anonymous) @ gui.js:354
gui.js:37 POST https://o95242.ingest.sentry.io/api/4504436020019200/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=a89e60810d444088ae7957d3a84837ee 403
```

</details>

xoniq, 2023-01-05:
I had to go back to 1.11.0 to make it work. 1.12.x didn't work also.

https://github.com/balena-io/etcher/releases/tag/v1.11.0

mcraa, 2023-01-05:
Probably #3891 fixes it as the source lib will be removed.

mcraa, 2023-01-19:
fixed in [v1.14.3](https://github.com/balena-io/etcher/releases/tag/v1.14.3)

closed 2023-01-19 by mcraa