balena-io/etcher #3969: help!

tampatattoojosh, 2023-01-09:
i downloaded "balenaEtcher" after prompted when i downloaded "tails" onto a flashdrive. now i cannot flash the "balenaEtcher" when prompted. what do i do, and do i even need "balenaEtcher" to run "tails" from flashdrive?

im new to this kind of thing, and do not know any code

mcraa, 2023-01-09:
There are many guides in many formats around the internet
https://www.youtube.com/watch?v=dH5Hm764hFU
https://opensource.com/article/18/7/getting-started-etcherio
Etcher and the site may look a bit different when they are older tutorials.

1. download the tails image
1. download (and install if not portable) etcher
1. open etcher
1. in etcher select the tails image for the first step, select the pendrive you want to write, click flash, wait to finish
1. boot from the written drive

lurch, 2023-01-09:
Note that you can't simultaneously read and write an image from the same flashdrive, so you need to copy the tails image onto a separate flashdrive (or harddrive) from the flashdrive you want to write the image to.