balena-io/etcher #3973: After starting flashing the Progress bar shows 'starting...' and abruptly stops. MacOs | Balena 1.13.1

Ranelkin, 2023-01-10:
- **Etcher version:**
- 1.13.1
- 
- **Operating system and architecture:** 
- macOs Monterey 12.6.2
- 64 bit
- 
- **Image flashed:** 
- kali-linux-2022.4-installer-amd64.iso
- 
- **What do you think should have happened:** 
- It should've normal flashed the iso image onto a external hard drive (also failed with normal usb)  
- 
- **What happened:**
- After selecting target and pressing 'flash' the progress startet showing 'starting...' and then abruptly stops 
- 
- **Do you see any meaningful error information in the DevTools?** 
- Information log: 
- Tue Jan 10 2023 23:58:54 GMT+0100 (Mitteleuropäische Normalzeit) Flash ({"image":{"size":3797270528,"name":"kali-linux-2022.4-installer-amd64.iso","hasMBR":true,"partitions":[{"offset":10233856,"size":540672,"type":239,"index":1}],"extension":"iso","path":"/Users/user/Downloads/kali-linux-2022.4-installer-amd64.iso"},"drives":[{"size":160041885696,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk3","error":null,"isReadOnly":false,"displayName":"/dev/disk3","blockSize":512,"isSCSI":false,"isRemovable":false,"device":"/dev/disk3","busVersion":null,"isSystem":false,"busType":"USB","isCard":false,"isUSB":true,"devicePath":"IODeviceTree:/PCI0@0/XHC1@14","mountpoints":[{"path":"/Volumes/Backup Festplatte","label":"Backup Festplatte"}],"description":"ASMT ASM105x Media","isUAS":null,"partitionTableType":"gpt"}],"driveCount":1,"uuid":"759c5733-20d1-43af-a505-da28413c1bdd","status":"started","flashInstanceUuid":"759c5733-20d1-43af-a505-da28413c1bdd","sample":0.1,"applicationSessionUuid":"58393ed0-37c4-4b51-91bb-c87c89360486","flashingWorkflowUuid":"433782b7-cd41-47ea-9638-69885b19d831"})
gui.js:37 Elevating command: /Applications/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper (Renderer).app/Contents/MacOS/balenaEtcher Helper (Renderer) /Applications/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js
_______________________________________________________________

- Error msg 1: 
gui.js:37 {"stack":"Error: Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'"}
console.<computed> @ gui.js:37
logException @ gui.js:37
(anonymous) @ gui.js:354
gui.js:37 POST https://o95242.ingest.sentry.io/api/4504436020019200/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=a89e60810d444088ae7957d3a84837ee 403
(anonymous) @ gui.js:37
_makeRequest @ gui.js:37
_sendProcessedPayload @ gui.js:37
_send @ gui.js:37
_processException @ gui.js:37
_handleStackInfo @ gui.js:37
captureException @ gui.js:37
_promiseRejectionHandler @ gui.js:37
gui.js:37 POST https://o95242.ingest.sentry.io/api/4504436020019200/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=a89e60810d444088ae7957d3a84837ee 403
_______________________________________________________________

Error msg 2: 
Failed to load resource: the server responded with a status of 403 ()                2 times
_______________________________________________________________

Error msg 3: 
gui.js:37 Uncaught RangeError: Maximum call stack size exceeded                   4 times
    at Function.isArray (<anonymous>)
    at gui.js:37
    at Array.forEach (<anonymous>)
    at step (gui.js:37)
    at gui.js:37
_______________________________________________________________

Error msg 4
gui.js:37 {"stack":"Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'"}
_______________________________________________________________

Error msg 5: 
Failed to load resource: the server responded with a status of 403 ()
_______________________________________________________________

I was executing with administrative rights. I also tried the same with a different smaller image of Kali Linux net installer and it didn't work. 
I also tried the same with my old Mac PC which runs on macOS Catalina and it had exactly the same problem. 

this is a screen recording i did 
https://user-images.githubusercontent.com/83548106/211682084-ec8a341a-67f5-4a84-9cae-b8695a894bed.mov

aethernet, 2023-01-11:
Thanks for the report! 
We're looking into it.

DimiDoukas, 2023-01-11:
I have the same kind of issue, first with Monterey and then also after I updated from Monterey to Mac OS Ventura 13.1., hoping that would fix the problem, but no. I'm trying to flash SD card with Home Assistant image for Raspberry https://github.com/home-assistant/operating-system/releases/download/9.4/haos_rpi4-64-9.4.img.xz . It lets me select the target and when I click flash button, it shows less than half a second the flash screen and comes right away back to the main screen, having emptied the target drive selection. I don't have access or knowledge to look any devtool so I can't see any error messages.

Ranelkin, 2023-01-12:
Update, it doesn't work with balena 1.12.0 either. It works however with balena 1.11.0

mcraa, 2023-01-12:
Thanks for the update, that really narrows it down to a problem with the i18n. :(

DimiDoukas, 2023-01-12:
Tried latest version 1.13.3 with Mc OS Ventura 13.1, still having the same problem that I had with 1.13.1, stopping right after I've pressed the 'flash' button and returning back to the main screen. With these errors

Thu Jan 12 2023 17:46:32 GMT+0200 TypeError: Converting circular structure to JSON
    --> starting at object with constructor 'Object'
    --- property 'top' closes the circle
gui.js:35  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:37 Terminating IPC server
gui.js:37 Flash results Object
gui.js:37 Thu Jan 12 2023 17:46:32 GMT+0200 Elevation cancelled ({"image":{"size":2147483648,"compressedSize":272991328,"hasMBR":true,"path":"https://github.com/home-assistant/operating-system/releases/download/9.4/haos_rpi3-64-9.4.img.xz","partitions":[{"offset":1048576,"size":33554432,"type":12,"index":1},{"offset":512,"size":1048064,"type":238,"index":2}],"isCompressed":true,"name":"1d363a73-0538-4e35-b57e-645c58a31bc9","isSizeEstimated":false,"extension":"xz"},"drives":[{"size":31927042048,"isVirtual":false,"enumerator":"DiskArbitration","logicalBlockSize":512,"raw":"/dev/rdisk4","error":null,"isReadOnly":false,"displayName":"/dev/disk4","blockSize":512,"isSCSI":false,"isRemovable":true,"device":"/dev/disk4","busVersion":null,"isSystem":false,"busType":"Secure Digital","isCard":true,"isUSB":false,"devicePath":"IODeviceTree:/arm-io/apcie@90000000/pci-bridge1@1/pcie-sdreader@0","mountpoints":[{"path":"/Volumes/PI","label":"PI"}],"description":"Apple SDXC Reader Media","isUAS":null,"partitionTableType":"mbr"}],"driveCount":1,"uuid":"975b6da5-736b-46e4-8cf1-2bb2613f1f17","status":"cancel","flashInstanceUuid":"975b6da5-736b-46e4-8cf1-2bb2613f1f17","sample":0.1,"applicationSessionUuid":"987c3bb0-1280-442d-ae7a-1c206d32f321","flashingWorkflowUuid":"a9ad9d63-8415-4350-a4df-5028f0f46cfd"})

{"stack":"Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'\n    at EventEmitter.i.invoke (electron/js2c/renderer_init.js:73:529)","message":"Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'"}
console.<computed> @ gui.js:37
o95242.ingest.sentry.io/api/4504486002884608/store/?sentry_version=7&sentry_client=raven-js%2F3.27.2&sentry_key=09ec9bc77ae643cf82b6fba244b149f2:1 

Failed to load resource: the server responded with a status of 403 ()

propercroc, 2023-01-14:
Sama here with mac os 13.0.1 (22A400) on Intel, workaround was using Balena 1.11.0.

closed 2023-01-19 by balena-ci

utk2103, 2023-11-05:
Etcher version:
1.18.11
Operating system and architecture:
Microsoft Windows 11 Home Single Language 
64 bit
Image flashed:
manjaro-kde-23.0.4-231015-linux65.iso
What happened:
After selecting target and pressing 'flash' the progress then after flashing, in validating process it shows error containing
Error: error invoking remote method 'enable-screensaver' : No handler registered for 'enable-screensaver'

EdjeElectronics, 2024-01-16:
Hey all, I was also getting the following error while trying to flash an image. balenaEtcher version is 1.18.11.

```
Error: error invoking remote method 'enable-screensaver' : No handler registered for 'enable-screensaver'
```

The error occurred because I was running balenaEtcher in Administrator mode on Windows. I was able to resolve the error by running in regular user mode instead.

PR #3982 (2023-01-19): patch: fixed mac sudo on other languages
fixes #3973