balena-io/etcher #3984: Website problems

lurch, 2023-01-20:
I just noticed that the https://www.balena.io/etcher has now swapped from a "dark theme" to a "light theme" - pretty!

However I also spotted that it says "Looking for Debina (.deb) packages" at the bottom, and "Debina" should obviously be "Debian" :joy: 

Also, that link goes to `https://github.com/balena-io/etcher?d_id=92e06341-01fc-4a38-a96b-16e28fc39aa8&s_id=1672760860908#debian-and-ubuntu-based-package-repository-gnulinux-x86x64` which changes to `https://github.com/balena-io/etcher?d_id=92e06341-01fc-4a38-a96b-16e28fc39aa8&s_id=1672760860908#debian-and-ubuntu-based-package-repository-gnulinux-x86x64&d_id=0fa62809-b13d-41cc-b9cc-89c15aa36b12R&s_id=1674212203308` when I actually click on it, and all the "extra stuff" in the URL means that the github page that gets displayed doesn't auto-scroll to the right part.
Whereas https://github.com/balena-io/etcher#debian-and-ubuntu-based-package-repository-gnulinux-x86x64 _does_ scroll to the right part of the page.

lurch, 2023-01-20:
Also, the FAQ part of that page talks about `ETCHER_FAKE_S3_LATEST_VERSION` in the "How can I simulate an update alert?" and IIRC that code was removed a long **long** time ago? (and doesn't get mentioned in https://github.com/balena-io/etcher/blob/master/FAQ.md )

Lizzieepton, 2023-01-20:
Hi @lurch - thank you for noticing the changes to the theme! We moved the site over to using Webflow to build this week and I really appreciate you letting us know about issues that happened during the migration, I just fixed the spelling and link issue from your first comment.

Lizzieepton, 2023-01-20:
Also you are correct, that FAQ is out of date, I have removed it and asked the Etcher team to review the rest of the FAQs on the website.

lurch, 2023-01-20:
Wow, thanks for the quick updates @Lizzieepton !
The "extra stuff in the URL" is also still present in other links on that page though e.g.
`https://github.com/balena-io/etcher?d_id=92e06341-01fc-4a38-a96b-16e28fc39aa8&s_id=1672760860908#redhat-rhel-and-fedora-based-package-repository-gnulinux-x86x64`
`https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md?d_id=258cd779-6616-480b-859a-1af708291d01R&s_id=1668158885391#why-is-my-drive-not-bootable`
`https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md?d_id=258cd779-6616-480b-859a-1af708291d01R&s_id=1668158885391#deactivate-desktop-shortcut-prompt-on-gnulinux`
`https://github.com/balena-io/etcher/blob/master/docs/USER-DOCUMENTATION.md?d_id=258cd779-6616-480b-859a-1af708291d01R&s_id=1668158885391#runtime-gnulinux-dependencies`
etc. etc.

I also just noticed that https://www.balena.io/etcher-pro probably needs some updates too:
 * "Now, we are working on a way for you to write to more devices" probably ought to be rephrased, if EtcherPro is now a shipping product?
 * FAQ says "We are taking pre-orders" and talks about a "Special Edition" but https://store.balena.io/products/etcher-pro? mentions neither of those
 * FAQ says $990 but store page says $1190

Lizzieepton, 2023-01-24:
Thanks for the heads up , I've now gone through all the links and got rid of the extra 'stuff' not sure where it was coming from! And fixed the FAQs on the Etcher-pro page and checking with the Etcher-Pro team about why the price is listed differently, I have a feeling it has something to do with tax but I agree it could be clearer :)

closed 2023-07-12 by dfunckt