balena-io/etcher #4008: Blank Page at Launch

Kobalt77, 2023-02-12:
- Etcher version: 1.14.3
- Operating system and architecture: Win7 x64
- Image flashed: none
- What happened: At launch, the Etcher shows me a blank page, no graphics or buttons or links whatsoever
[image]

mcraa, 2023-02-13:
Hi, can you add the logs from the devtools (press Ctrl + Shift + I) console?
Are you sure you are running x64?

In the meantime you may use an older version like [1.7.9](https://github.com/balena-io/etcher/releases/download/v1.7.9/balenaEtcher-Portable-1.7.9.exe), link also available on the [official site](https://www.balena.io/etcher#download-etcher).

Kobalt77, 2023-02-13:
Hi, if you're talking about my Windows 7, yes I'm sure that I'm running a 64bit version but for the Etcher, I found no real .exe for x64, maybe my mistake?

there is some more screenshots
[image]
[image]

mcraa, 2023-02-14:
Thanks, the logs about the `usb` module will make it easier to find the problem.

KevinGroninga, 2023-04-27:
I too am having the very same issue with a Win 7, x64. Using versions 'balenaEtcher-Setup-1.18.4.exe' and 'balenaEtcher-Setup-1.18.8.exe'.  Starts up, nothing but a blank window.

[image]

z1ad, 2023-09-19:
It seems to have stopped working on v1.7.0+ for Windows 7
Ive downloaded v1.6.0 (or older) and can confirm it loads fine
You can find old releases here: https://github.com/balena-io/etcher/releases

KevinGroninga, 2023-09-19:
Thanks!  I found, downloaded, and installed v1.6.0.  Seems to work again!

al20878, 2024-02-21:
JFYI, Etcher 1.18.11 (the latest as of today) is still all blank on Windows-7 64 Home...  If it is no longer supposed to be run on Windows-7 the installer should warn and refuse to install, rather than installing the application, which is broken / useless.

Kalle-M-P, 2024-03-22:
Another data point on this strange issue.  
I am using **Win8.1Pro-32bit** and wanting to make a linux install USB

I had two versions of Rufus on my machine and an upgrade was offered that I declined because it wanted to remove my existing version, now both have vanished and I am weary of Rufus.  

So I try Balena/etcher again
I downloaded **balenaEtcher-Portable-1.7.9.exe** because I am on 32bit
 and remember that the same happened before.  Blank screen.

Here is the error from the Developer tools console, I learnt something new about snooping on JavaScript applications **(press Ctrl + Shift + I)**, thank you @mcraa .  This is above my pay grade but perhaps it will offer a clue to someone smarter than me.

I downloaded earlier version **balenaEtcher-Portable-1.6.0.exe** as suggested by @z1ad and the GUI is visible.

I have not done recent development and found out recently that pretty much all dev tools require 64bit OS.  So I may have very old .NET and other libraries and toolchains on my machine, was hoping to do some development on the 64bit-linux install now under work.

It seems to have flashed and verified the 32GB Kingston USB memory stick.
GRUB >> Mint-live-linux >> Starting install >> Installation Complete, no error:05 like Rufus USB >> booting of HDD

=====
```
index.css:1 
Failed to load resource: net::ERR_FILE_NOT_FOUND

electron/js2c/asar_bundle.js:5 
Uncaught Error: \\?\C:\Users\Kalle\AppData\Local\Temp\2..........c\resources\app\generated\modules\@ronomon\direct-io\binding.node is not a valid Win32 application.
\\?\C:\Users\Kalle\AppData\Local\Temp\2..........c\resources\app\generated\modules\@ronomon\direct-io\binding.node
    at process.func [as dlopen] (electron/js2c/asar_bundle.js:5)
    at Object.Module._extensions..node (internal/modules/cjs/loader.js:1138)
    at Object.func [as .node] (electron/js2c/asar_bundle.js:5)
    at Module.load (internal/modules/cjs/loader.js:935)
    at Module._load (internal/modules/cjs/loader.js:776)
    at Function.f._load (electron/js2c/asar_bundle.js:5)
    at Function.o._load (electron/js2c/renderer_init.js:33)
    at Module.require (internal/modules/cjs/loader.js:959)
    at require (internal/modules/cjs/helpers.js:88)
    at Object.<anonymous> (gui.js:1)
```
=====

MissLavender-LQ, 2024-04-02:
getting same issue on arch appimage and aur