balena-io/etcher #4014: Etcher for macOS should not direct ask for user password

DominikHoffmann, 2023-02-17:
- **Etcher version: 1.14.3** 
- **Operating system and architecture: macOS Ventura 13.2; Apple Silicon** 
- **Image flashed: pfSense-plus-compat-recovery-23.01-RELEASE-aarch64.img** 
- **What do you think should have happened: When getting ready to flash, the app should have presented an OS dialog asking for an admin user ID and password.** 
- **What happened: Instead, it presented its own dialog and only asked for a password. This makes it suspicious for looking like a password phishing tool. **
- **Do you see any meaningful error information in the DevTools?**

lurch, 2023-02-17:
Duplicate of e.g. #3025 , #3321 , etc. ?

mcraa, 2023-02-17:
Thanks for your feedback, closing as duplicate.
No PRs were made from the previous issues, but since then the package we use got forked, updated many times, so we may be able find a better solution.

closed 2023-02-17 by mcraa