balena-io/etcher #4015: setup.rpm.sh asks for yum-utils, which fails on F37

opohorel, 2023-02-17:
Tested on Fedora 37

On line 521 of setup.rpm.sh script, there is a `check_rpm_tool "yum-utils"`.
"yum-utils" can't be currently installed on F37, as it is a retired package.
There is a "dnf-utils" package that provides the same functionality.

Simply switching to `check_rpm_tool "dnf-utils"`, makes this script work flawlessly.

Please consider changing condition to something like:
```

    if test "$manager" == "yum"; then
        check_rpm_tool "yum-utils"
    fi

    if test "$manager" == "dnf"; then
        check_rpm_tool "dnf-plugins-core"
    fi
```
"dnf-plugins-core" provides both "dnf-utils" and "dnf-plugin-config-manager"

mcraa, 2023-02-17:
Thanks for the feedback, and the proposed solution
as shown in the readme 
>[image] \
Package repository hosting is graciously provided by  [Cloudsmith](https://cloudsmith.com)

 the setup sh is maintained by them.
If you can raise the issue to them that would be helpful, we will check and ping them as well.

opohorel, 2023-02-17:
I pinged them on their support mail. Thanks for pointing it out

opohorel, 2023-03-13:
Cloudsmith has resolved this issue now with an updated setup.rpm.sh

closed 2023-03-13 by opohorel