balena-io/etcher #4016: Unsupported distro help

intresti, 2023-02-17:
Im using Nobara 37.

When trying to get etcher packages i get a distro unsupported error (Im new to linux)

This is the terminal:

Reading package lists... Done
Building dependency tree... Done
E: Unable to locate package balena-etcher-electron
[j@nobara-pc opt]$ sudo apt install -y balena-etcher-electroncurl -1sLf \
  'https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh' \
  | sudo -E bash
E: Command line option '1' [from -1sLf] is not understood in combination with the other options.
[j@nobara-pc opt]$ curl -1sLf \
  'https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh' \
  | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'rpm' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=nobara  version=37  codename=Thirty  arch=x86_64  
 >>>>:
   OK: Importing 'balena/etcher' repository GPG keys into rpm ...
   OK: Checking for available package manager (DNF/Microdnf/YUM/Zypper) ...
 ^^^^: ... Detected package manager as 'dnf'
 NOPE: Checking for dnf dependency 'yum-utils' ...
   OK: Attempting to install 'yum-utils' ...
 NOPE: Checking for dnf dependency 'dnf-plugin-config-manager' ...
   OK: Attempting to install 'dnf-plugin-config-manager' ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=nobara  version=37  codename=Thirty  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as SUSE Linux Enterprise Server 12, use the following:
 >>>>:
 >>>>:   <curl command> | distro=sles version=12.1 codename= sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_CkgBh6swP (might not be errors, nor even relevant):
 !!!!:
 !!!!: > Last metadata expiration check: 0:40:13 ago on Fri 17 Feb 2023 12:30:19
 !!!!: > GMT.
 !!!!: > Package dnf-plugins-core-4.3.1-2.fc37.noarch is already installed.
 !!!!: > Dependencies resolved.
 !!!!: > Nothing to do.
 !!!!: > Complete!

mcraa, 2023-02-17:
Hi,

the error message tells you what to do. 
```
You can force this script to use a particular value by specifying distro,
: version, or codename via environment variable. E.g., to specify a distro
: such as SUSE Linux Enterprise Server 12, use the following:
:
: | distro=sles version=12.1 codename= sudo -E bash
```
It is pretty difficult for Cloudsmith or anyone else to keep up with all the different distro names.
As a quick search shows Nobara is based on Fedora, and Fedora is supported.
So you can force the script to think you are on fedora and it should work.
```
curl -1sLf \
   'https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh' \
   | distro=fedora version=37 codename= sudo -E bash
```

closed 2023-02-17 by mcraa

intresti, 2023-02-17:
Thank you! 

Can you please tell me how to use etcher after i have the repository?