balena-io/etcher #4022: bz2-compressed img files get stuck at extremely slow flashing speed.

TDuffinNTU, 2023-03-05:
- **Etcher version:** 
1.18.4 x64

- **Operating system and architecture:** 
Linux x64 Appimage (SteamOS/Arch based and KDE Neon/Ubuntu Based)

- **Image flashed:** 
SteamOS recovery partition (compressed with bz2).
https://store.steampowered.com/steamos/download/?ver=steamdeck&snr=

- **What do you think should have happened:** 
Balena either should have refused to use my disk image, or it should've flashed the USB at regular speed.

Uncompressing the image with my OS, and then flashing that uncompressed file, worked fine with a regular flashing speed.

- **What happened:**
Appeared to start flashing, but claimed 0.20MB/s speed, 10+ hour ETA, and effectively wrote nothing to the USB. No warnings or errors even in the console.

- **Do you see any meaningful error information in the DevTools?** 

I tried this, but nothing out of order.

I can see that bz2 compressed images have caused other issues which are still open. If bz2 is going to cause issues into the future (especially with certain OS maintainers packaging ISOs in the format and recommending etcher too) I would suggest it's marked as an unsupported or problematic format. It would've solved me a lot of time assuming my USB or other setup was the issue when it was actually the fault of the compression.

lurch, 2023-03-06:
Maybe related to #3901 which also mentioned SteamOS ?

TDuffinNTU, 2023-03-06:
Looks similar. Though couldn't find error message in the console in my own case.

aethernet, 2023-03-09:
Let me check that

aethernet, 2023-03-09:
I reproduced your issue, and will investigate further

dfunckt, 2023-07-12:
Dupe of #4038

closed 2023-07-12 by dfunckt