balena-io/etcher #4026: macOS: Etcher should work from non-privileged account and ask for admin user ID and password

DominikHoffmann, 2023-03-09:
- **Etcher version:**  1.14.3
- **Operating system and architecture:** macOS 13.2.1, Apple Silicon M1 MAX
- **Image flashed:** fpSense recovery firmware
- **What do you think should have happened:** Etcher should have asked me for not just admin password, but also the admin user ID and used the proper API to do so.
- **What happened:** Etcher asked for just the admin password through its own dialog, raising the suspicion that it is an admin password phishing operation. This also means that one cannot use it from a non-privileged account. I usually am not logged into my admin account, and having to copy the image file over to the admin account, switching accounts and then flashing is more inconvenient than it has to be.
- **Do you see any meaningful error information in the DevTools?** N/A

lurch, 2023-03-15:
See also #3798 and #3321

aethernet, 2023-03-24:
I agree 100%

It bothers me as well.

It's on my list of things to look at, unfortunately, It's not trivial and I don't have the time atm.

PR welcome :)

kmbgit, 2023-06-15:
This works from a terminal (where "admin" is the name of a user with Administrator permission)...

su admin -c "sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher"

Alternatively, download the Raspberry Pi Imager from https://www.raspberrypi.com/software/ - it manages to ask for an admin username/password so it must be possible.

dfunckt, 2023-07-12:
Closing in favor of #3321

closed 2023-07-12 by dfunckt