balena-io/etcher #4029: I am not able to install balena-etcher in OpenSuse Tembleweed via terminal or via yast

serial02, 2023-03-14:
Hey guys

 I am not able to install balena-etcher in OpenSuse Tembleweed via terminal or via yast.
 The error message appears that the header of the application is not signed, see image below.

 The problem occurs in any version of balena-etcher

```
Error: INVALID:balena-etcher-1.18.4-1.x86_64 (file-60781681): Signature verification failed [6-The file is not signed]
     Header SHA256 digest: OK
     Header SHA1 digest: OK
     Payload SHA256 digest: OK
     MD5 digestion: OK
     The packet header is not signed!
```

[image]

serial02, 2023-03-16:
Hey guys

 I was able to install balena-etcher via official balena-etcher repository on openSuse Tumbleweed with this tip below:

So you need following command to add the repo:

`curl -1sLf 'https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh'   | codename="tumbleweed" sudo -E bash`

Then:

```
sudo zypper up
sudo zypper install balena-etcher
```

However, the installed version is 1.14.3 while the latest version is 1.18.5.
 So, can you give an estimate of when the balena-etcher repositories will be updated with the new version 1.18.5 via the balena-etcher repository for Open Suse Tumbleed?

balena-etcher-1.14.3 is from January 2023. Version 1.18.4 was released just 2 weeks ago on github.  All intermediate versions are pre-releases only.  Since developers use an external cloud service to host their build repository, it takes some time for the repository to be updated, correct?

[image]

serial02, 2023-03-19:
> edson@localhost:~> sudo sudo zypper -vv ref && sudo zypper -vvv dup
> [sudo] password for root:
> Details: 3
> initializing target
> Specified repositories:
> Checking whether to update metadata for balena-etcher
> Problem downloading files from 'balena-etcher'.
> Download (curl) error for 'https://dl.cloudsmith.io/public/balena/etcher/rpm/opensuse/tumbleweed/x86_64/repodata/repomd.xml':
> Error code: HTTP response: 402
> Error message: The requested URL returned error: 402
> 
> See the error message above for a tip.
> Ignoring repository 'balena-etcher' due to above error.
> Checking whether to update metadata for balena-etcher-noarch
> Problem downloading files from 'balena-etcher-noarch'.
> Download (curl) error for 'https://dl.cloudsmith.io/public/balena/etcher/rpm/opensuse/tumbleweed/noarch/repodata/repomd.xml':
> Error code: HTTP response: 402
> Error message: The requested URL returned error: 402
> 
> See the error message above for a tip.
> Ignoring repository 'balena-etcher-noarch' due to above error.
> Checking if it is necessary to update the metadata for balena-etcher-source
> Problem downloading files from 'balena-etcher-source'.
> Download (curl) error for 'https://dl.cloudsmith.io/public/balena/etcher/rpm/opensuse/tumbleweed/SRPMS/repodata/repomd.xml':
> Error code: HTTP response: 402
> Error message: The requested URL returned error: 402
> 
> See the error message above for a tip.
> Ignoring repository 'balena-etcher-source' due to above error.
> Checking if you need to update metadata for Brave Browser
> Downloading: https://brave-browser-rpm-release.s3.brave.com/[completed (3.3 KiB/s)]
> The 'Brave Browser' repository is up to date.
> Checking whether to update metadata for Core Repository (Non-OSS)
> Downloading: http://download.opensuse.org/tumbleweed/repo/n[done (10.1 KiB/s)]
> Repository 'Main Repository (Non-OSS)' is up to date.
> Checking whether to update metadata for Master Repository (OSS)
> Downloading: http://download.opensuse.org/tumbleweed/repo/o[completed (10.8 KiB/s)]
> The repository 'Main Repository (OSS)' is up to date.
> Checking whether to update metadata for Update Master Repository
> Downloading: http://download.opensuse.org/update/tumbleweed/[completed (2.9 KiB/s)]
> Repository 'Update Master Repository' is up to date.
> Skipping disabled repository 'openSUSE-20230310-0'
> Checking whether to update metadata for packman
> Downloading: https://ftp.gwdg.de/pub/linux/misc/packman/suse[completed (1.3 KiB/s)]
> The 'packman' repository is up to date.
> Bypassing disabled repository 'openSUSE-Tumbleweed-Debug'
> Checking if you need to update metadata for Open H.264 Codec (openSUSE Tumbleweed)
> Downloading: http://codecs.opensuse.org/openh264/openSUSE_Tu[completed (1.7 KiB/s)]
> The 'Open H.264 Codec (openSUSE Tumbleweed)' repository is up to date.
> Bypassing disabled repository 'openSUSE-Tumbleweed-Source'
> Some of the repositories were not updated due to an error.
> edson@localhost:~>

[image]

galagithub, 2023-04-26:
seems to be related to #4057 and #3511 and #4034

aethernet, 2023-04-27:
Related indeed. Here's what i posted there for reference: 

Hello,

We're moving out of Cloudsmith as continuing to host the package there would be too costly for us.

We're very thankful for the service and help CloudSmith provided, and would recommend their product for smaller open source app and private packages anytime!

At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self hosted (probably CloudFlare R2 + workers) in the upcoming weeks.

So sorry for the inconvenience and thanks you for your patience.

Note that you can always get a .deb and .rpm package in the release section of the github repository.

closed 2023-04-27 by aethernet

TobiPeterG, 2024-01-04:
Hey there,

I created an OBS repo that pulls the official RPM and creates a new package with it. Kind of dirty, but the easiest way. This way, dependencies should be recognized as well.
However, the package isn't really compiled against openSUSE libs, so issues might occur (though it works fine for me).

The link to the repo:
https://download.opensuse.org/repositories/home:/Tobi_Peter:/balena-etcher/openSUSE_Tumbleweed/

Add it via zypper:
`sudo zypper ar -f -p 90 https://download.opensuse.org/repositories/home:/Tobi_Peter:/balena-etcher/openSUSE_Tumbleweed/ etcher`

I'll try to keep it up to date, it's still on 1.18.11 due to a bug I noticed