balena-io/etcher #4034: Cloudsmith repository gives an HTTP 402 status

ask-compu, 2023-03-19:
apparently 402 means payment required?

MarkyMarkDE, 2023-03-19:
exact the same here, the source is no more signed (sends 402 "Payment Required") and disabled by my system.
But when i go to Updates / Other Software, i can't find this source (PPA) - really suspicious.

[image]

Marcos-Gallardo, 2023-03-19:
Same here, fresh Ubuntu 22.04.2 install...

ElScotto, 2023-03-19:
Same for the YUM/DNF (aka "rpm") instances.
It's definitely affecting the entire suite of "etcher" repositories hosted by Cloudsmith.

For people using yum/dnf, add "`skip_if_unavailable=True`" to each stanza in the "`/etc/yum.repos.d/balena-etcher.repo`" file.  This will at least allow yum/dnf checks and updates to operate whilst the balena-etcher repository is out of commission.  I've attached my amended repo configuration file here for reference.
A manual attempt to `wget` the setup file mentioned in the install notes returns (as others have found) "402 - Payment Required"

```
$ wget https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh
--2023-03-20 09:12:25--  https://dl.cloudsmith.io/public/balena/etcher/setup.rpm.sh
Resolving dl.cloudsmith.io (dl.cloudsmith.io)... 18.67.111.69, 18.67.111.67, 18.67.111.84, ...
Connecting to dl.cloudsmith.io (dl.cloudsmith.io)|18.67.111.69|:443... connected.
HTTP request sent, awaiting response... 402 Payment Required
2023-03-20 09:12:26 ERROR 402: Payment Required.
```
[balena-etcher.repo.txt](https://github.com/balena-io/etcher/files/11012435/balena-etcher.repo.txt)
[balena-etcher.update.fail.txt](https://github.com/balena-io/etcher/files/11012447/balena-etcher.update.fail.txt)

shreefgit, 2023-03-20:
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/kinetic/InRelease  402  Payment Required [IP: 18.67.111.84 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu kinetic InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.

Thouareg, 2023-03-20:
> E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/kinetic/InRelease 402 Payment Required [IP: 18.67.111.84 443] E: The repository [...]

Same here too

tomstoneham, 2023-03-20:
Related from last year: #3672

Might be worth moving to an alternative host for deb repos.

jdrch, 2023-03-20:
> > E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/kinetic/InRelease 402 Payment Required [IP: 18.67.111.84 443] E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu kinetic InRelease' is no longer signed. N: Updating from such a repository can't be done securely, and is therefore disabled by default.
> 
> Same here too, but for debian repo

Same here on Debian Bullseye

jdrch, 2023-03-20:
> Related from last year: #3672
> 
> Might be worth moving to an alternative host for deb repos.

I'd agree, except that this is unfortunately PFTC for many packages sourced from 3rd party repos. At least once per year someone forgets to update or renew something and the repo breaks 🤦‍♂️

sn: good find on that issue; I don't even remember posting it 😂

drbornot, 2023-03-20:
Same here:
```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/pop/dists/jammy/InRelease  402  Payment Required [IP: 143.204.89.12 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/pop jammy InRelease' is no longer signed.
```

TomerSchDev, 2023-03-20:
Also here:
Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/linuxmint/dists/vanessa/InRelease  402  Payment Required [IP: 13.226.2.75 443]The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/linuxmint vanessa InRelease' is no longer signed.

jdrch, 2023-03-20:
> When I do the curl download of the setup script with verbose output it says this: "x-detail: Access to the repository has been restricted."

So even initial setup is failing. Yikes.

ToasterUwU, 2023-03-20:
Same Here. Why use Cloudsmith anyways? Saw some old issues with the same error info and it seems like cloudsmith requires payment based on amount of downloads. 

Why not make a launchpad PPA which doesnt have this issue?

jdrch, 2023-03-20:
> Why not make a launchpad PPA which doesnt have this issue?

FWIW I've seen PPAs break too. Typically because someone forgot to renew the cert and so the PPA became unsigned. Different failure mode, same end user UX.

TomerSchDev, 2023-03-20:
> > Why not make a launchpad PPA which doesnt have this issue?
> 
> FWIW I've seen PPAs break too. Typically because someone forgot to renew the cert and so the PPA became unsigned. Different failure mode, same end user UX.

Is there something that can bw donw to prevent this? I can see it's not the firat time that this is happning and probbley not the last

jdrch, 2023-03-20:
> Is there something that can bw donw to prevent this? I can see it's not the firat time that this is happning and probbley not the last

It's generally up to package dev and/or maintainer orgs to manage their corresponding repos.

ToasterUwU, 2023-03-20:
> FWIW I've seen PPAs break too. Typically because someone forgot to renew the cert and so the PPA became unsigned. Different failure mode, same end user UX.

Yes, but these are all issues that can be prevented with some maintance effor. The repo breaking down because the max download per month limit was reached is neither something that is easily predictable, and also not easily fixable.
If a video goes viral that brings lots of people to download Etcher, it could be easily broken for the rest of the month, unless they pay extra, in which case it takes money and time until the repo works again.
If they would use a Launchpad PPA, they have no issue like this. They can make a little list of things to make sure and if they do it right, they wont run risk of outages for more than a few minutes here and there. With cloudsmith it can break for days and up to almost an entire month.

No solution is perfect, but this current solution is definetly not the best one by far.

jdrch, 2023-03-21:
@ToasterUwU Can PPAs be used to host .rpm packages too? Stupid question, but I'm not sure I've seen that before. I believe I have Etcher on my openSUSE installation; wouldn't want to see that disappear. 

While we're on the topic, openSUSE's Open Build Service 3rd party packages for multiple distros and distro types. Could be another option.

ToasterUwU, 2023-03-21:
> @ToasterUwU Can PPAs be used to host .rpm packages too? Stupid question, but I'm not sure I've seen that before. I believe I have Etcher on my openSUSE installation; wouldn't want to see that [...]

@jdrch fair point. I also have never seen this being used for FeDora based systems. So it's possible this doesn't work. But for those systems there is most certainly something like launchpad.
I mean you can host a repo yourself very easily if you want to as well. With that you don't have to pay per download either, just need a server to run this on, and I'm sure Belena has more than one server which could do this.

jdrch, 2023-03-21:
> But for those systems there is most certainly something like launchpad.

Maintaining separate repos would increase dev overhead and therefore the likelihood of other more serious errors/outages. Ditto running your own repo server. From my observation, it seems repos generally need someone whose only job is to look after them or bad things happen. I doubt the team has the manpower for that. The repos of larger, for profit orgs like TeamViewer and Microsoft have similar incidents at least once per year, too (I know due to using them).

ElScotto, 2023-03-21:
It seems to be a regular problem as previously mentioned by @ToasterUwU.
I imagine that if Balena are going to stay with Cloudsmith, they will need to upgrade their plan once more, as they did in Jan 2022 (cf "[https://forums.balena.io/t/balenaetcher-deb-repo-requires-payment/350765](https://forums.balena.io/t/balenaetcher-deb-repo-requires-payment/350765)" and "[https://forums.balena.io/t/i-have-a-problem-when-apply-apt-get-update/350767/1](https://forums.balena.io/t/i-have-a-problem-when-apply-apt-get-update/350767/1)")

jdrch, 2023-03-21:
Given the lack of dev team attention to this issue, I've reported it on the [official forums](https://forums.balena.io/t/balenaetcher-deb-repo-requires-payment/350765/18?u=jdrch) in the same thread as last year's incident to emphasize that this is a recurring issue. If anyone else wants to chime in to help it get more traction, I'm sure we'd all appreciate it.

bstivers, 2023-03-21:
I understand this isn't a dev team issue, but I'm going to mini rant here anyway,

Why would anyone pay for Etcher Pro, when they can't even get the free version? Not a good first impression. I've literally installed thousands of different Linux programs a whole lot more popular, and I've absolutely never had this issue with a single piece of software. Yet this is a recurring problem? Aren't software developers, business owners, etc., supposed to learn from their previous mistakes?

Is this their own way of limiting how much profit they make on Etcher Pro?

What's the Pro account money going towards if they can't even keep a simple repo up and working? How much are they paying Cloudsmith now? 26GB/mo is literally less than $100/month. I bet I can host it on my Oracle Cloud account for a heckuva lot cheaper....  

Also, 250GB/mo is $700. Are they really going over that? Doubtful. That's literally only 2.5 Pilot Belena subscriptions. Half of the cost of a single Production subscription.  You're telling me they don't have 1 single Production subscription or over 2 Pilots? C'mon now. I understand developing software isn't easy. But paying bills when you have a paid platform is.

aethernet, 2023-03-21:
Hello

We're actively looking at a solution. 

If you must know, the threshold is 2TB and been a couple days before renewal.

We're looking at ways to increase this limit, which is not as simple as it sounds.

If you have any recommendation for a hosting provided for open source software without such limitation, please share.

Worse case scenario it would be reset by same time tomorrow.

ElScotto, 2023-03-21:
> > Hello
> > We're actively looking at a solution.
> > If you must know, the threshold is 2TB and has been hit a couple days before renewal.
> > We're looking at ways to increase this limit, which is not as simple as it sounds.
> > If you have any recommendation for a hosting provided for open source software without such limitation, please share.
> > Worse case scenario it would be reset by same time tomorrow.
> 
> @aethernet is it not possible to host on launchpad? Please feel free to read this: https://help.launchpad.net/legal I'm not sure, if Balena can fulfill this rules, but if then it is free of charge But I think this is the most common way for Linux

FYI @MarkyMarkDE , the actual URL for Launchpad's "Legal" page is [https://help.launchpad.net/Legal](https://help.launchpad.net/Legal); the site's URLs are case-sensitive.

So, Launchpad definitely supports PPAs for Ubuntu-based distros, but there are wider use-cases than just Ubuntu and its derivatives.
I can't find anything specific about it, so I'm throwing it out here; "_Can RPM repositories be served from Launchpad too?_"
Given that Launchpad seems closely allied with Canonical & Ubuntu, my gut feeling is "**no**".  So, if that's the case, what other options are available for RPM & PPA repositories that are cost-effective for projects like Balena-etcher?

MarkyMarkDE, 2023-03-21:
> FYI @MarkyMarkDE , the actual URL for Launchpad's "Legal" page is https://help.launchpad.net/Legal; the site's URLs are case-sensitive.
>
> So, Launchpad definitely supports PPAs for Ubuntu-based [...]

@ElScotto sorry, i have hovered the URL in the Footer on the main page and here is "legal" (lower case written) ...
Isn't it possible here on Git?

ElScotto, 2023-03-21:
> @ElScotto sorry, i have hovered the URL in the Footer on the main page and here is "legal" (lower case written) ... Isn't it possible here on Git?

If you go to the "legal" link rather than the "Legal" link, Launchpad brings up this text:
**This page does not exist yet. You can create a new empty page, or use one of the page templates.**
I've also captured the page and pasted it here...
Anyway, this bit's getting off-topic, so we should get back to the main issue; Cloudsmith vs Launchpad vs ??? for hosting open-source project package repositories.

[image]

MarkyMarkDE, 2023-03-21:
> If you go to the "legal" link rather than the "Legal" link, Launchpad brings up this text: **This page does not exist yet. You can create a new empty page, or use one of the page templates.** I've [...]

@ElScotto yes, this is really off-topic, so we go to my last question
>>Isn't it possible here on Git?
[image]

ElScotto, 2023-03-21:
Aha! Found the discrepancy in the URLs.  Your original link is from "**help**.launchpad.net", the other one from your pasted picture is for "launchpad.net"; different subdomains and pages!

So, back to repository hosts, I would presume that there's no **technical** difficulty in hosting a PPA and RPM repository on github.  The main issue will be ensuring that the repository file structure and necessary metadata files are present and maintained and that the various package update commands (eg `yast`, `dnf`, `yum`, `apt`/`apt-get` etc) can reliably pull from a github URL link to the appropriate repository head.

jdrch, 2023-03-22:
I haven't been able to find anything that indicates Launchpad supports non-DEB distros. As such, my preference is for [Open Build Service](https://openbuildservice.org/) where, AFAIK, the only thing you have to worry about it keeping the repo signed.

I've seen others mention GitHub; personally I've never used a GitHub repo with a package manager so I'm not sure how that works.

paspo, 2023-03-22:
> We're looking at ways to increase this limit, which is not as simple as it sounds.
> 
> If you have any recommendation for a hosting provided for open source software without such limitation, please share.

Think about [Cloudflare R2](https://www.cloudflare.com/products/r2/) which is an object storage (S3) with no paid egress.
At the end of the day, a package repo is all about static data, so making use of a company with a strong CDN makes sense.

Cloudsmith is clearly not a viable solution anymore (unless you can get a deal in some form); in fact, I don't understand these limits: they're ok for smaller open-source project, but when you grow you generate more bandwidth, so IMHO it should be better for Cloudsmith to "support" a known project instead of "yet another javascript framework used by 2 people"...

aethernet, 2023-03-22:
Hello everyone,

Short term: 10 hours ago CloudSmith lifted the bandwidth limit for this period so everything should be working again. The period renew in ~2h so we have close to a month to find a proper solution.

Mid/long term we're evaluating different options.
I personally like github and/or cloudflare options as we're already using both (and from a maintenance perspective I prefer not to introduce more tools/services).

Thanks for the very good suggestion here, very much appreciated!

MarkyMarkDE, 2023-03-22:
> Hello everyone,
>
> Short term: 10 hours ago CloudSmith lifted the bandwidth limit for this period so everything should be working again. The period renew in ~2h so we have close to a month to find a [...]

@aethernet yes, the Problem is currently gone! Thanks!

[image]

jdrch, 2023-03-22:
> I personally like github and/or cloudflare options as we're already using both (and from a maintenance perspective I prefer not to introduce more tools/services).

Assuming those support DEB, RPM, and the usual Windows options too, I'm all for 'em. Glad things are working again :)

lskillen, 2023-03-23:
Hey folks, Lee from Cloudsmith here. 👋 First of all, apologies for the outage; even when it is a customer of ours hitting limits, it's still painful enough that we try to reach out to get it sorted as fast as possible, including bumping those limits temporarily, no questions asked, to get people back to a working state; in fact, we hate any pipelines being broken for any reason. :)

As we advance, it's up to the Balena team to decide which approach to take, but it's important to state that we provide much more value than just a static store for binary packages. If Balena were interested, they could distribute securely from custom domains worldwide, using Balena-based keys/signatures backed by an availability well beyond what GitHub or others offer.

Trust me, I _love_ GitHub, and we use it _a lot_, but that pink unicorn rears its beautiful head way more often than any of us would like. We have customers that prefer us because of it. We're _built for this exact purpose_ as an all-in-one solution that takes care of global scaling and distribution of all package formats (plus the possibility of private channels, paid ESAs, trust controls, etc.)

We're happy to support open-source and the community _for free_ where possible, but please remember that as a start-up, we also have our bills/vendors to pay for. So far, there's been _no cost_ to Balena, and they're using 10x our standard amount for OSS. So we need to cap open-source usage at some level, especially when customers are themselves commercial, for-profit entities.

Typically we'll work with those types of entities to establish a commercial relationship or enter into a significant quid pro quo sponsorship package. Again, this is Balena's decision, not ours, and we'll support them no matter their choice of long-term solutions; I know and trust they'll have the community in mind regardless.

ask-compu, 2023-03-23:
> Hey folks, Lee from Cloudsmith here. wave First of all, apologies for the outage; even when it is a customer of ours hitting limits, it's still painful enough that we try to reach out to get it sorted as fast as possible, including bumping those limits temporarily, no questions asked, to get people back to a working state; in fact, we hate any pipelines being broken for any reason. :)
> 
> As we advance, it's up to the Balena team to decide which approach to take, but it's important to state that we provide much more value than just a static store for binary packages. If Balena were interested, they could distribute securely from custom domains worldwide, using Balena-based keys/signatures backed by an availability well beyond what GitHub or others offer.
> 
> Trust me, I _love_ GitHub, and we use it _a lot_, but that pink unicorn rears its beautiful head way more often than any of us would like. We have customers that prefer us because of it. We're _built for this exact purpose_ as an all-in-one solution that takes care of global scaling and distribution of all package formats (plus the possibility of private channels, paid ESAs, trust controls, etc.)
> 
> We're happy to support open-source and the community _for free_ where possible, but please remember that as a start-up, we also have our bills/vendors to pay for. So far, there's been _no cost_ to Balena, and they're using 10x our standard amount for OSS. So we need to cap open-source usage at some level, especially when customers are themselves commercial, for-profit entities.
> 
> Typically we'll work with those types of entities to establish a commercial relationship or enter into a significant quid pro quo sponsorship package. Again, this is Balena's decision, not ours, and we'll support them no matter their choice of long-term solutions; I know and trust they'll have the community in mind regardless.

oi a github issue is not the place to try and advertise for ur company

lskillen, 2023-03-23:
> oi a github issue is not the place to try and advertise for ur company

It's not an advertisement, it's an explanation as to what happened (and an apology on our part, we should have noticed it sooner and reached out to Balena quicker). They're literally using us to distribute the software that caused this issue.

tomstoneham, 2023-03-23:
@lskillen I totally understand your business' concerns, and understand why you impose monthly transfer limits on projects such as this. My problem with Cloudsmith is the way that the issue here is passed on to end users. When the Etcher repo returns a 4xx status code, it doesn't just prevent etcher updates/installations, but also any `apt update`/`apt upgrade` on machines which have the repo in their `sources.list` (i.e: anyone who has previously installed Etcher). So updates for all software, including critical security updates, can't be installed until the repo is manually removed, or Balena increases their usage limit.

I don't know much about the internals of apt, but is it possible to disable access in a way that doesn't break all updates on end-user machines? Perhaps allow fetching `Release`/`InRelease` but not actual build artifacts, so that machines with Etcher already installed can see there's no updates, and don't try to fetch binaries, and new installations can't go ahead? I imagine the transfer cost of meta files like this is much lower than Etcher's binaries.

ask-compu, 2023-03-23:
> @lskillen I totally understand your business' concerns, and understand why you impose monthly transfer limits on projects such as this. My problem with Cloudsmith is the way that the issue here is [...]

what? apt does not block updates when a repository is broken, it just won't update packages from the broken repository

tomstoneham, 2023-03-23:
@ask-compu Ah, scratch that then. I just removed the repo immediately and was going off (apparently faulty) memory. Sorry @lskillen!

lskillen, 2023-03-23:
> is it possible to disable access in a way that doesn't break all updates on end-user machines

I think you do have a fair point, @tomstoneham, but just different from the `apt` issue - I can see above that we block the installation script as well when it happens, and I don't think that's right, so that might be something to fix at least.

We _did_ experiment with trying to rate-limit downloads instead of turning them off completely, which would be a lot "softer" of a fail state, but unfortunately, technical constraints prevented it. It's still on our roadmap, though!

For bandwidth usage, we do have notifications that we send out. I wonder if they got missed here, or if they went to the wrong (maybe old) email addresses, but those are there to try and help people mitigate issues (limits) before they happen. 🤔

ElScotto, 2023-03-24:
> @ask-compu Ah, scratch that then. I just removed the repo immediately and was going off (apparently faulty) memory. Sorry @lskillen!

@tomstoneham, you're quite correct to ask the question about the potential for failed/unavailable repositories blocking package updates.
For RPM distros that use `yum` &/or `dnf` (I don't know about `zypper`), if a repository is unavailable for whatever reason, it can indeed block package updates (including security updates) **unless** "`skip_if_unavailable=True`" is added to each stanza in the relevant repository `.repo` file ("`/etc/yum.repos.d/balena-etcher.repo`" in this particular instance).
I mentioned this earlier - refer to https://github.com/balena-io/etcher/issues/4034#issuecomment-1475418183

If `skip_if_unavailable` is not defined **at all**, `dnf`/`yum` will default it to **`false`**, which means a failure to access _any_ repository for whatever reason will see the entire `update` or `check-update` operation fail.  In other words, repositories that have the potential for being unavailable _must_ have this option set, otherwise updates and checks will fail even for packages provided by the core/base system repositories.

MarkyMarkDE, 2023-04-16:
since today, the problem is back again. Cloudsmith sends 402 "Payment required" Status

MarkyMarkDE, 2023-04-21:
> Hello
>
> We're actively looking at a solution.
>
> If you must know, the threshold is 2TB and has been hit a couple days before renewal.
>
> We're looking at ways to increase this limit, which is not as [...]

you asked for an alternate hoster? Maybe gitlab.com?
Here have some linux programs their ppa's ...

jdrch, 2023-04-23:
So it looks like we've gotten to the point where this product is available to download only part of the month and the repo breaks system updates during that time. What a sad state of affairs.

lskillen, 2023-04-25:
> So it looks like we've gotten to the point where this product is available to download only part of the month and the repo breaks system updates during that time it's unavailable, as if the former situation wasn't bad enough. What a sad state of affairs.

To be clear, Cloudsmith bumped the allocation up to a further higher level (15x the standard amount, up from 10x), while Balena decided on what to do. Cloudsmith also implemented the changes previously discussed such that:

1. Installing the repository itself will allow config and GPG key fetches even if bandwidth has run out
2. `skip_if_unavailable=true` is now present in new pulls of config so it _doesn't_ break system updates.

However, I believe they have now made the decision to discontinue the use of the apt/rpm repositories (with Cloudsmith, at least, they may have an alternative), so things are out of our hands now. Obviously, this is a tough situation, because bandwidth usage is out of their control, and there's pressure on both sides to pay bills.

So, I just wanted to thank Balena and their team for their patronage, and for everyone here who used Cloudsmith or repositories hosted there for the past few years. I'm sorry that we couldn't fully work it out in this case. I'm still a fan of the Balena products, personally. :) All the best to all of you folks!

aethernet, 2023-04-25:
Hello, 

We're moving out of Cloudsmith as continuing to host the package there would too costly for us.

We're very thankful for the service and help CloudSmith provided, and would recommend their product for smaller open source app and private packages.

At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self hosted (probably CloudFlare R2 + workers)

So sorry for the inconvenience and thanks you for your patience.

galagithub, 2023-04-26:
just opened #4057 for the same thing !

jdrch, 2023-04-26:
Does anyone know (or can you link to) how to `skip_if_unavailable` for `zypper` & `apt`?

> At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self hosted (probably CloudFlare R2 + workers) in the upcoming weeks.

@aethernet Thanks for the info! What's your recommended course of action for those of us who currently have the Cloudsmith repo installed and for whom many jobs are broken due to that? Should we remove the repo, config our package manager to skip it, purge the package entirely, or something else?

ElScotto, 2023-04-26:
> Does anyone know (or can you link to) how to set up `skip_if_unavailable` or its equivalent for `zypper` & `apt`?
> 
> > At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self hosted (probably CloudFlare R2 + workers) in the upcoming weeks.
> 
> @aethernet Thanks for the info! What's your recommended course of action for those of us who currently have the Cloudsmith repo installed and for whom many jobs are broken due to that? Should we remove the repo, config our package manager to skip it, purge the package entirely, or something else?

@jdrch, I don't think `zypper` or `apt` (currently) support anything like the "`skip_if_unavailable=True`" option for `dnf` & `yum`.
A quick search shows that this is a feature requested for zypper (**[openSUSE/libzypp/issue/369](https://github.com/openSUSE/libzypp/issues/369#issue-1089337341)**), although **[this other link](https://github.com/openSUSE/zypper/issues/19#issuecomment-15163165)** indicates that `zypper-1.8.13` does allow one to permanently disable an unreachable repository after using the `ignore` option.
Further searches indicate that the only option for `apt` is to manually remove the PPA; there appears to be no option to temporarily bypass an unavailable repository/PPA.
**_Please note, I do not run Debian/Ubuntu derivatives nor do I run any form of SUSE Linux, so my comments are based purely on internet search results._**

lurch, 2023-04-26:
> What's your recommended course of action for those of us who currently have the Cloudsmith repo installed and for whom many jobs are broken due to that?

@aethernet I guess there's going to be _many_ people in this boat, so maybe it's worth creating a sticky issue at https://github.com/balena-io/etcher/issues ? :thinking:

jdrch, 2023-04-26:
@ElScotto TYVM!

@lurch Yep, we're in our 2nd breakage in as many months, after all.

aethernet, 2023-04-27:
@lurch good call, just did

aethernet, 2023-04-27:
@ElScotto 

> Should we remove the repo

Yes, and probably get a new version from the GH releases page.

closed 2023-04-27 by aethernet

jall0330, 2023-05-06:
Here is how I solved:

1. Issue doing a sudo apt update (Debian 11)
/opt/balenaEtcher$ sudo apt update
Hit:1 https://ocean.surfshark.com/debian stretch InRelease
Hit:3 https://brave-browser-apt-release.s3.brave.com stable InRelease                                                                                         
Hit:4 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                                                                     
Hit:5 https://dl.winehq.org/wine-builds/debian bullseye InRelease                                                                                                                                                 
Hit:6 http://deb.debian.org/debian bullseye InRelease                                                                                                                                                             
Hit:7 http://security.debian.org/debian-security bullseye-security InRelease                                                                                                       
Hit:2 https://mxrepo.com/mx/repo bullseye InRelease                                                                                                                                
Hit:8 http://ftp.us.debian.org/debian bullseye InRelease                                                                        
Err:9 https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease                         
  402  Payment Required [IP: 18.155.248.62 443]
Hit:10 http://deb.debian.org/debian bullseye-updates InRelease                                            
Hit:11 http://deb.debian.org/debian bullseye-backports InRelease                    
Hit:12 http://ftp.us.debian.org/debian bullseye-updates InRelease                   
Get:13 https://mega.nz/linux/repo/Debian_11 ./ InRelease [2,959 B]
Reading package lists... Done         
N: Skipping acquire of configured file 'main/binary-i386/Packages' as repository 'https://brave-browser-apt-release.s3.brave.com stable InRelease' doesn't support architecture 'i386'
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/debian/dists/bullseye/InRelease  402  Payment Required [IP: 18.155.248.62 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye InRelease' is not signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.

2. Fixed first issues with architecture i386
3. using synaptic uninstalled Balena Etcher version 1.7.9
4. Downloaded latest version 1.18. from git hub
5. installed using dpkg -i balena-etcher_1.18.4_amd64.deb
6. then navigate to /etc/apt/sources.list.d  do a ls -ltr
7. ls
root@debjall0330:/etc/apt/sources.list.d# ls -ltr
total 28
-rw-r--r-- 1 root root 201 Jan 29 20:23 google-chrome.list
-rw-r--r-- 1 root root   1 Jan 29 20:23 spotify.list
-rw-r--r-- 1 root root 165 Jan 29 20:23 mx.list
-rw-r--r-- 1 root root  52 Feb  8 20:17 surfshark.list
-rw-r--r-- 1 root root 104 Apr 27 13:45 megasync.list
-**rw-r--r-- 1 root root 460 May  6 14:00 balena-etcher.list**
-rw-r--r-- 1 root root 135 May  6 14:01 brave-browser-release.list

9. using gedit / nano /vim or any other tool load the balena etcher source list that is giving issue
10. for example: nano balena-etcher-list
11. # Source: balena
# Site: https://github.com/balena-io/etcher
# Repository: balena / etcher
# Description: Flash OS images to SD cards & USB drives, safely and easily.

# deb [signed-by=/usr/share/keyrings/balena-etcher-archive-keyring.gpg] https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye main

# deb-src [signed-by=/usr/share/keyrings/balena-etcher-archive-keyring.gpg] https://dl.cloudsmith.io/public/balena/etcher/deb/debian bullseye main

Mark both lines then save
12. then sudo apt update
13. sudo apt update
[sudo] password for jorge: 
Hit:1 https://ocean.surfshark.com/debian stretch InRelease
Hit:3 http://deb.debian.org/debian bullseye InRelease                                                                                                                                                             
Hit:4 http://security.debian.org/debian-security bullseye-security InRelease                                                                                             
Hit:5 https://dl.google.com/linux/chrome/deb stable InRelease                                                                                                            
Hit:6 http://deb.debian.org/debian bullseye-updates InRelease                                                                                                           
Hit:7 http://ftp.us.debian.org/debian bullseye InRelease                                                                               
Hit:8 https://dl.winehq.org/wine-builds/debian bullseye InRelease                                               
Hit:9 http://deb.debian.org/debian bullseye-backports InRelease                                                 
Hit:10 http://ftp.us.debian.org/debian bullseye-updates InRelease                         
Hit:2 https://mxrepo.com/mx/repo bullseye InRelease                                       
Get:11 https://mega.nz/linux/repo/Debian_11 ./ InRelease [2,959 B]
Fetched 2,959 B in 2s (1,837 B/s)     
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
All packages are up to date.

Issue is fixed and balena etcher lastest version is installed

jdrch, 2023-05-07:
@jall0330 Thanks, but I already purged the package & removed the repo from all my machines.

MarkyMarkDE, 2023-05-31:
Is the repository at cloudsmith online again? Since yesterday, I not get the status "payment required" back, seems that the update has already finished successfully ...

jdrch, 2023-05-31:
> Is the repository at cloudsmith online again? Since yesterday, I not get the status "payment required" back, seems that the update has already finished successfully ...

The dev told us either here or in another issue that the cloudsmith repo is effectively dead.

lurch, 2023-05-31:
> The dev told us either here or in another issue that the cloudsmith repo is effectively dead.

Yep, see #4059

MarkyMarkDE, 2023-05-31:
> > The dev told us either here or in another issue that the cloudsmith repo is effectively dead.
> 
> Yep, see #4059

okay,  have now removed `sudo unlink /etc/apt/sources.list.d/balena-etcher.list` from my `/etc/apt/sources.list.d` Folder and updated my ppa repos `sudo apt-get update` and now the entry is gone - but it's hilarious, the ppa is down, but it gives NOT an error back.

yahyayossef, 2023-06-06:
All errors are solved

[2 more replies]