balena-io/etcher #4038: Can't flash on Monterey v12 on M1 mac with v1.18.4

lukedds, 2023-03-23:
- **Etcher version:** 1.18.4
- **Operating system and architecture:** M1 Mac Monterey 12 to a Sandisk 32Gb USB device
- **Image flashed:** steamdeck recovery image
- **What do you think should have happened:** should have flashed the steamdeck image to the device
- **What happened:** Never gets past 'Flashing 0%', rolled back to 1.14.3 and it works fine
- **Do you see any meaningful error information in the DevTools?** Just stays on 0% with an unfathomably long ETA, no actual errors

dfunckt, 2023-03-23:
I'm on Monterey on Mac M1 and I can flash just fine. Can you share a link to the image you're flashing so that I can give it a go?

lukedds, 2023-03-23:
No worries, thank you, happy to retest any changes.

aethernet, 2023-03-24:
https://github.com/balena-io/etcher/issues/4022

tuiliere, 2023-04-05:
Reproduced same problem : 1.18.4 was unable to flash from url (home assistant rasp version :  ) on M1 Macbookpro  OSX ventura 13.2.1, while 1.14.3 does it fine.

In case... dev console logs for 1.18.4 failure follows :
Failed to load resource: net::ERR_FILE_NOT_FOUND
gui.js:354  _____ _       _
|  ___| |     | |
| |__ | |_ ___| |__   ___ _ __
|  __|| __/ __| '_ \ / _ \ '__|
| |___| || (__| | | |  __/ |
\____/ \__\___|_| |_|\___|_|

Interested in joining the Etcher team?
Drop us a line at join+etcher@balena.io

Version = 1.18.4, Type = dmg
4gui.js:1 Uncaught Error: aborted
    at connResetException (node:internal/errors:691:14)
    at TLSSocket.socketCloseListener (node:_http_client:402:19)
    at TLSSocket.emit (node:events:538:35)
    at node:net:687:12
    at TCP.done (node:_tls_wrap:580:7)
gui.js:37 Elevating command: /Users/jerome/Desktop/balenaEtcher.app/Contents/Frameworks/balenaEtcher Helper (Renderer).app/Contents/MacOS/balenaEtcher Helper (Renderer) /Users/jerome/Desktop/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js
gui.js:37 Error: Error invoking remote method 'disable-screensaver': No handler registered for 'disable-screensaver'
    at o.invoke (node:electron/js2c/renderer_init:57:526)
(anonymous) @ gui.js:37
gui.js:35  0 devices, 0% at 0.00 MB/s (total 0.00 MB/s)  with 0 failed devices
gui.js:37 Successfully connected to IPC server: etcher-server-3664, socket root /var/folders/49/_lp9qx4j1nx06f46gj3_t9780000gn/T/
gui.js:37 Image: https://github.com/home-assistant/operating-system/releases/download/9.5/haos_rpi4-64-9.5.img.xz
gui.js:37 Devices: /dev/disk5
gui.js:37 Auto blockmapping: true
gui.js:37 Decompress first: true
gui.js:37 Terminating IPC server
gui.js:37 Flash results Object
gui.js:37 Error: /Users/jerome/Desktop/balenaEtcher.app/Contents/Resources/app/generated/child-writer.js:1
(()=>{var __webpack_modules__=[,(__unused_webpack_module,exports,__webpack_require__)=>{"use strict";Object.defineProperty(exports,"__esModule",{value:true});const tslib_1=__webpack_require__(2);tslib_1.__exportStar(__webpack_require__(3),exports);tslib_1.__exportStar(__webpack_require__(114),exports);tslib_1.__exportStar(__webpack_require__(122),exports);tslib_1.__exportStar(__webpack_require__(118),exports);tslib_1.__exportStar(__webpack_require__(149),exports);tslib_1.__exportStar(__webpack_require__(179),exports);tslib_1.__exportStar(__webpack_require__(180),exports);tslib_1.__exportStar(__webpack_require__(68),exports);tslib_1.__exportStar(__webpack_require__(181),exports);tslib_1.__exportStar(__webpack_require__(234),exports);tslib_1.__exportStar(__webpack_require__(235),exports);tslib_1.__exportStar(__webpack_require__(28),exports);tslib_1.__exportStar(__webpack_require__(236),exports);tslib_1.__exportStar(__webpack_require__(293),exports);tslib_1.__exportStar(__webpack_require__(69),exports);tslib_1.__exportStar(__webpack_require__(121),exports);tslib_1.__exportStar(__webpack_require__(322),exports);tslib_1.__exportStar(__webpack_require__(323),exports);tslib_1.__exportStar(__webpack_require__(324),exports);tslib_1.__exportStar(__webpack_require__(325),exports);tslib_1.__exportStar(__webpack_require__(259),exports)},(__unused_webpack_module,__webpack_exports__,__webpack_require__)=>{"use strict";__webpack_require__.r(__webpack_exports__);__webpack_require__.d(__webpack_exports__,{__assign:()=>__assign,__asyncDelegator:()=>__asyncDelegator,__asyncGenerator:()=>__asyncGenerator,__asyncValues:()=>__asyncValues,__await:()=>__await,__awaiter:()=>__awaiter,__classPrivateFieldGet:()=>__classPrivateFieldGet,__classPrivateFieldIn:()=>__classPrivateFieldIn,__classPrivateFieldSet:()=>__classPrivateFieldSet,__createBinding:()=>__createBinding,__decorate:()=>__decorate,__exportStar:()=>__exportStar,__extends:()=>__extends,__generator:()=>__generator,__importDefault:()=>__importDefault,__importStar:()=>__importStar,__makeTemplateObject:()=>__makeTemplateObject,__metadata:()=>__metadata,__param:()=>__param,__read:()=>__read,__rest:()=>__rest,__spread:()=>__spread,__spreadArray:()=>__spreadArray,__spreadArrays:()=>__spreadArrays,__values:()=>__values});var extendStatics=function(d,b){extendStatics=Object.setPrototypeOf||{__proto__:[]}instanceof Array&&function(d,b){d.__proto__=b}||function(d,b){for(var p in b)if(Object.prototype.hasOwnProperty.call(b,p))d[p]=b[p]};return extendStatics(d,b)};function __extends(d,b){if(typeof b!=="function"&&b!==null)throw new TypeError("Class extends value "+String(b)+" is not a constructor or null");extendStatics(d,b);function __(){this.constructor=d}d.prototype=b===null?Object.create(b):(__.prototype=b.prototype,new __)}var __assign=function(){__assign=Object.assign||function __assign(t){for(var s,i=1,n=arguments.length;i<n;i++){s=arguments[i];for(var p in s)if(Object.prototype.hasOwnProperty.call(s,p))t[p]=s[p]}return t};return __assign.apply(this,arguments)};function __rest(s,e){var t={};for(var p in s)if(Object.prototype.hasOwnProperty.call(s,p)&&e.indexOf(p)<0)t[p]=s[p];if(s!=null&&typeof Object.getOwnPropertySymbols==="function")for(var i=0,p=Object.getOwnPropertySymbols(s);i<p.length;i++){if(e.indexOf(p[i])<0&&Object.prototype.propertyIsEnumerable.call(s,p[i]))t[p[i]]=s[p[i]]}return t}function __decorate(decorators,target,key,desc){var c=arguments.length,r=c<3?target:desc===null?desc=Object.getOwnPropertyDescriptor(target,key):desc,d;if(typeof Reflect==="object"&&typeof Reflect.decorate==="function")r=Reflect.decorate(decorators,target,key,desc);else [... cut ...] (q[0][2],r)}function fulfill(value){resume("next
    at Object.createError (gui.js:37:219904)
    at elevateScriptCatalina (gui.js:341:4060)
    at async withTmpFile (gui.js:1:718820)
    at async elevateCommand (gui.js:341:4411)
    at async Server.<anonymous> (gui.js:341:8515)
(anonymous) @ gui.js:37
gui.js:37 Error: Error invoking remote method 'enable-screensaver': No handler registered for 'enable-screensaver'
    at o.invoke (node:electron/js2c/renderer_init:57:526)
(anonymous) @ gui.js:37

ryanburnett, 2023-04-06:
Same issue here - trying to flash the [Steam Deck OS](https://store.steampowered.com/steamos/download/?ver=steamdeck&snr=) with balenaEtcher 1.18.4 on macOS Ventura 13.3.

Granted balenaEtcher Full Disk Access and Files & Folders permissions and opened it with:

> sudo /Applications/balenaEtcher.app/Contents/MacOS/balenaetcher

Stuck on "Flashing 0%".

hellopahe, 2023-08-18:
I had the same issue, tried 1.17.0 and went through, hope this could help you.

DiracSpace, 2023-09-15:
I fixed the issue with Balena Etcher not flashing [steamdeck-recovery-4.img.bz2](https://store.steampowered.com/steamos/download/?ver=steamdeck&snr=) by doing the following steps.

After reading the following reply, I did the following to decompress the bzip2 file.

> if you extract the image and try to flash that it pretty fast

```
➜  Downloads bzip2 -d steamdeck-recovery-4.img.bz2
➜  Downloads ls -la | grep "steamdeck"
.rw-rw-r-- roberto roberto 7.2 GB Thu Sep 14 12:26:51 2023 steamdeck-recovery-4.img
```

It took a while but it eventually decompressed the file leaving only the image file. I'm not sure if Balena Etcher should be decompressing it automatically (and if you ask me I don't think it should be doing that as the software should only be flashing the image instead of also decompressing) but doing this extra step helps a lot.

[image]

SplitGemini, 2023-09-19:
It was found that when the decompressed image is larger than 5g, etcher will use the method of decompressing and burning at the same time. At this time, the burning will fail (for example, steamdeck.img). However, if the image is smaller than 5g, it will be decompressed first and then burned, and it will not fail.

SplitGemini, 2023-09-19:
Are there any relevant updates?

mbulut-domino, 2024-07-11:
Any updates on this??
Also had to downgrade to 1.14.3 but it only works in every full moon...

I don't understand why such serious bug in a generally great product is not prioritized higher? There are numerous duplicates of this issue on the tracker but it seems like they don't get the attention they should. :( 

Meanwhile, most of my Windows user colleagues have settled back on Rufus, which might not look that pretty but at least it works.