balena-io/etcher #4046: App Crashes on macOS Ventura

GerardoGarciaR, 2023-04-10:
**Etcher version:** 1.18.4
**Operating System:** macOS Ventura

**App (BalenaEtcher) crashes on macOS Ventura**

Yes, of course, the same error happens to me, in macOS version.
The app crashes at start, so I decided to know whats going on, and in order to know whats happening, I ran it on a Terminal (Shell), and thats what it say:

```
gera@Mac-Pro-de-Gera ~ % /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
objc[1526]: Class WebSwapCGLLayer is implemented in both /System/Library/Frameworks/WebKit.framework/Versions/A/Frameworks/WebCore.framework/Versions/A/Frameworks/libANGLE-shared.dylib (0x7ffb575e5378) and /Applications/balenaEtcher.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libGLESv2.dylib (0x111faece0). One of the two will be used. Which one is undefined.
[0410/113234.939059:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113235.101064:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113235.121999:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113235.305488:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113235.467395:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113235.679832:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113235.842255:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113235.856915:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113235.861612:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
objc[1540]: Class WebSwapCGLLayer is implemented in both /System/Library/Frameworks/WebKit.framework/Versions/A/Frameworks/WebCore.framework/Versions/A/Frameworks/libANGLE-shared.dylib (0x7ffb575e5378) and /Applications/balenaEtcher.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libGLESv2.dylib (0x110d86ce0). One of the two will be used. Which one is undefined.
[0410/113236.028570:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113236.045590:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113236.483796:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113236.646008:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113236.659989:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113236.663706:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113236.831080:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113236.849437:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
objc[1543]: Class WebSwapCGLLayer is implemented in both /System/Library/Frameworks/WebKit.framework/Versions/A/Frameworks/WebCore.framework/Versions/A/Frameworks/libANGLE-shared.dylib (0x7ffb575e5378) and /Applications/balenaEtcher.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libGLESv2.dylib (0x10c58bce0). One of the two will be used. Which one is undefined.
[0410/113237.026974:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113237.188800:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113237.206391:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113237.561448:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113237.727839:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113237.742041:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113237.746550:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
objc[1545]: Class WebSwapCGLLayer is implemented in both /System/Library/Frameworks/WebKit.framework/Versions/A/Frameworks/WebCore.framework/Versions/A/Frameworks/libANGLE-shared.dylib (0x7ffb575e5378) and /Applications/balenaEtcher.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libGLESv2.dylib (0x10a7a0ce0). One of the two will be used. Which one is undefined.
[0410/113237.909744:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113237.927096:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113238.294779:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113238.457604:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113238.471760:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113238.476892:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
objc[1547]: Class WebSwapCGLLayer is implemented in both /System/Library/Frameworks/WebKit.framework/Versions/A/Frameworks/WebCore.framework/Versions/A/Frameworks/libANGLE-shared.dylib (0x7ffb575e5378) and /Applications/balenaEtcher.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libGLESv2.dylib (0x113064ce0). One of the two will be used. Which one is undefined.
[0410/113238.640244:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113238.658145:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113239.024605:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113239.187888:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113239.202587:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113239.207587:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
objc[1549]: Class WebSwapCGLLayer is implemented in both /System/Library/Frameworks/WebKit.framework/Versions/A/Frameworks/WebCore.framework/Versions/A/Frameworks/libANGLE-shared.dylib (0x7ffb575e5378) and /Applications/balenaEtcher.app/Contents/Frameworks/Electron Framework.framework/Versions/A/Libraries/libGLESv2.dylib (0x10f2fdce0). One of the two will be used. Which one is undefined.
[0410/113239.371055:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113239.387665:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113239.759906:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113239.924301:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113239.938805:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113239.943940:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
[1551:0410/113240.060840:ERROR:gpu_init.cc(481)] Passthrough is not supported, GL is disabled, ANGLE is 
[0410/113240.106119:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113240.123732:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113240.484652:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113240.646000:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113240.659481:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113240.663167:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
[1555:0410/113240.780670:ERROR:gpu_init.cc(481)] Passthrough is not supported, GL is disabled, ANGLE is 
[0410/113240.824310:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[1523:0410/113240.841375:ERROR:network_service_instance_impl.cc(470)] Network service crashed, restarting service.
[0410/113241.191168:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[0410/113241.352276:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
[0410/113241.365906:WARNING:exception_types.cc(171)] EXC_CRASH should not contain exception 12
[1523:0410/113241.369501:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
[... cut ...]
[1523:0410/113242.076495:ERROR:gpu_process_host.cc(974)] GPU process exited unexpectedly: exit_code=9
[1523:0410/113242.076535:FATAL:gpu_data_manager_impl_private.cc(450)] GPU process isn't usable. Goodbye.
[0410/113242.233192:WARNING:exception_snapshot_mac.cc(70)] exception EXC_GUARD invalid in EXC_CRASH
zsh: trace trap  /Applications/balenaEtcher.app/Contents/MacOS/balenaEtcher
```

But the curious thing is that the same App runs fine on macOS BigSur on the same machine.
But It doesn't work on Ventura.

My GPU is an NVIDIA GT710 1GB.

aethernet, 2023-04-26:
Thanks for the report.

If you're on the nvidia gpu you're on mac intel.
I don't have such a setup available for testing, so I might require your help for this

Karan1458, 2023-06-06:
+1 on this, Facing same issue. 
OS: Ventura
GPU: GT710 2gb

dfunckt, 2023-07-12:
macOS has long dropped support for nVidia graphics cards, so it's not surprising you're facing issues, and I'm afraid the error comes from within Chromium (GPU and network related files show up in the logs above) which we have zero ability to modify. I'm tentatively closing the issue, feel free to reopen if you don't agree with my conclusion and have further info to debug.

closed 2023-07-12 by dfunckt