balena-io/etcher #4048: Unable to install on latest Debian based version of Linux Mint

MrBrailey, 2023-04-14:
- **Etcher version: 1.18.4** 
- **Operating system and architecture: x64** 
- **Image flashed:balenaEtcher-1.18.4-x64.AppImage** 
- **What do you think should have happened:** Since this is a Debian version of Linux Mint, and I have run Etcher on Debian 12 I thought it would install without a hitch. Attached is the text from the terminal when it gronked.
- **What happened:**Installation failed!
- **Do you see any meaningful error information in the DevTools?** Here's what popped up on the terminal window.
- ephraim@development:~$ curl -1sLf \
   'https://dl.cloudsmith.io/public/balena/etcher/setup.deb.sh' \
   | sudo -E bash
Executing the  setup script for the 'balena/etcher' repository ...

   OK: Checking for required executable 'curl' ...
   OK: Checking for required executable 'apt-get' ...
   OK: Detecting your OS distribution and release using system methods ...
 ^^^^: ... Detected/provided for your OS/distribution, version and architecture:
 >>>>:
 >>>>: ... distro=linuxmint  version=5  codename=elsie  arch=x86_64  
 >>>>:
   OK: Checking for apt dependency 'apt-transport-https' ...
   OK: Checking for apt dependency 'ca-certificates' ...
   OK: Checking for apt dependency 'gnupg' ...
   OK: Checking for apt signed-by key support ...
   OK: Importing 'balena/etcher' repository GPG key ...
 NOPE: Checking if upstream install config is OK ...
 ^^^^: ... Failed to fetch configuration for your OS distribution release/version.
 >>>>:
 >>>>: It looks like we don't currently support your distribution release and
 >>>>: version. This is something that we can fix by adding it to our list of
 >>>>: supported versions (see contact us below), or you can manually override
 >>>>: the values below to an equivalent distribution that we do support:
 >>>>:
 >>>>: ... distro=linuxmint  version=5  codename=elsie  arch=x86_64  
 >>>>:
 >>>>: You can force this script to use a particular value by specifying distro,
 >>>>: version, or codename via environment variable. E.g., to specify a distro
 >>>>: such as Ubuntu/Xenial (16.04), use the following:
 >>>>:
 >>>>:   <curl command> | distro=ubuntu version=16.04 codename=xenial sudo -E bash
 >>>>:
 !!!!: Oh no, your setup failed! :-( ... But we might be able to help. :-)
 !!!!: 
 !!!!: You can contact balena for further assistance.
 !!!!: 
 !!!!: URL: https://github.com/balena-io/etcher
 !!!!: 
 !!!!: Last 20 log lines from .csm_setup_dIklcftUl (might not be errors, nor even relevant):
 !!!!:
 !!!!: > ii  gnupg                                         2.2.27-2+deb11u2
 !!!!: > all          GNU privacy guard - a free PGP replacement
 !!!!: > ii  gnupg-l10n                                    2.2.27-2+deb11u2
 !!!!: > all          GNU privacy guard - localization files
 !!!!: > ii  gnupg-utils                                   2.2.27-2+deb11u2
 !!!!: > amd64        GNU privacy guard - utility programs
 !!!!: > ii  python3-gnupg                                 0.4.6-1
 !!!!: > all          Python wrapper for the GNU Privacy Guard (Python 3.x)

MrBrailey, 2023-04-14:
Ok, I moved the balenaEtcher-1.18.4-x64.AppImage to my desktop, changed the permissions to make it executable and it looks like **the dang thing is working**. The Debian install instructions though still fail as before, but it will take a more seasoned Debian geek to decipher the error messages. Thanks for creating a great USB creation program.

aethernet, 2023-04-25:
Sorry for the inconvenience. We're moving out of cloudsmith atm. The docs will be updated in the upcoming minutes.

closed 2023-04-25 by aethernet