balena-io/etcher #4050: App is losing path name of "device to write to", after "Flash"-command

michael5411, 2023-04-16:
- **Etcher version:**    is 1.18.4
- **Operating system and architecture:** MacOS Ventura

1) I select a valid an tested URL to flash from
2) Select the target
3) Usd "Flash-Button"; balena-advertisement shows up ....
4) Password-window shows up an correct psw is entered
5) jumps back to the "select drive"-window and nothing happens.

There was a quite similar bug some years ago on exactly the same point. Balena fixed it. But now, it is not working.

huijsh, 2023-04-16:
same experience

aethernet, 2023-04-25:
Is there anything useful shown in the console? ( command + caps + I )

huijsh, 2023-04-25:
its the laptop. On the desktop it works fine. Thanks for reaching out.

dfunckt, 2023-07-13:
Can you please give v1.18.11 a go and see if the issue persists?

cmccall-epilog, 2024-03-07:
I'm also experiencing this on macOS 14 with 1.18.11 and what is currently on the master branch. Flashing progresses until it just stops logging progress. No errors in the console log, it just stops at "Flashing 9.48 GB" with the last console log output being: "Flashing 1 device, undefined% at 63.30 MB/s (total 63.30 MB/s)  with 0 failed devices" It seems to be stopping on some final step.