balena-io/etcher #4051: Etcher claims the freshly formatted target drive is the source drive and wont let me select it

alinanorakari, 2023-04-16:
- **Etcher version:** 1.18.4
- **Operating system and architecture:** macOS Ventura 13.3.1 on Intel iMac Pro
- **Image flashed:** self-built installation ISO for NixOS
- **What do you think should have happened:** I should have been able to select the thumb drive as my target drive to flash the ISO
- **What happened:** Etcher shows a blue "Source drive" label next to the thumb drive and disabled the check box to select it as the target
- **Do you see any meaningful error information in the DevTools?** no

The target selection dialog:

[image]

The location of the image:

[image]

Unusual things about the system:
- The boot drive is not the internal drive but an SSD connected via USB because the internal drive is flaky. The boot drive is `disk6` according to Disk Utility
   - `disk7` is the APFS container
   - `disk7s1s1` is the macOS volume
      - This is where the ISO resides that I want to flash
- The internal drive that is not used for booting is `disk0` ⚠️ Etcher misidentified this drive as the System drive
- The thumb drive got assigned `disk2`
   - There are no files on the thumb drive, it has just been formated and is empty

aethernet, 2023-04-25:
Looks like the same bug indeed.

that-ben, 2024-02-27:
A year later, this still happens on the most stable build.  Any progress on this?

aethernet, 2024-03-21:
Haven't been able to repro yet, so no progress

that-ben, 2024-04-03:
Honestly, this is quite a pain to do that every time, especially when your system volume is small.  I feel like Balena Etcher doesn't care about its macOS user base, which is quite sad.  Anyway, I just took a habit of launching Terminal and executing this and it works for me: sudo dd if=/dev/souce of=/dev/target

aethernet, 2024-06-07:
> Honestly, this is quite a pain to do that every time, especially when your system volume is small. I feel like Balena Etcher doesn't care about its macOS user base, which is quite sad. Anyway, I just took a habit of launching Terminal and executing this and it works for me: sudo dd if=/dev/souce of=/dev/target
> 
> Good luck everyone.

All the current maintainers are mac users :)
So trust me when I say we do care about mac ...

The problem here is that we have no repro...

6R1MR34P3R, 2024-07-12:
Happening the same on last update MacOS 15 Sequoia

aravikusu, 2024-08-08:
Was encountering this issue as well. I'm unsure if this will help any, but I was plugging my USB into a hub. When I tried plugging it into the machine directly it no longer incorrectly said that my USB was my "source drive", and let me select it.

oivinde, 2024-08-17:
> Was encountering this issue as well. I'm unsure if this will help any, but I was plugging my USB into a hub. When I tried plugging it into the machine directly it no longer incorrectly said that my USB was my "source drive", and let me select it.

I have a Mac Mini M2 with a docking base. Experience the same issue and solution. Plugged into dock, USB identifies as source, directly in Mac Mini, no problem. Drive is 64GB USB 3 SanDisk.

nocsi, 2025-01-08:
Ran into this issue today. Tried writing a 1TB image stored on an external drive to a target device - both devices connected via a thunderbolt dock. I suspect it has something to do with the USB bus hierarchy. For the record, I'm using the following:

* Mac Mini M4
* Caldigit Thunderbolt4 Dock
  * Onboard SDXC Port
  * Sony MRW-G2 Thunderbolt SD Card Reader
* 1TB MicroSD

What doesn't work:
* Target Drive in onboard SD Card reader with source image on external drive (Same Source)
* Target Drive through an external SD Card with source image on separate external drive (Same Source)

What does work:
* Target drive on a completely separate USB bus than the thunderbolt dock's. Attaching external SD Card reader directly to mac mini thunderbolt port.

There's probably some logic somewhere that assumes drives under a common hub are of the same source.

[image]
[image]

kneutron, 2025-02-02:
I just hit this bug on Sonoma 14.7.2, Intel 2018 mac mini, booting from external SSD SK Hynix Beetle 1TB

Source is ISO (see link below) in ZFS dataset on the beetle, target is a 64GB SDcard on a GVRIOY USB3 (usb-c) dock.

https://github.com/nchevsky/systemrescue-zfs/releases

https://www.amazon.com/gp/product/B0CLDDJ7LF?ie=UTF8&psc=1

Never seen this before, but Etcher is useless in this situation and I had to burn the ISO on another Mac.

yousihy, 2025-04-29:
I am not sure if this is going to help anyone. But I got the same error and I was stuck.

Then I tried something which actually worked!

I noticed that whenever I place the OS Image on any external drive or partition on my Mac, it will give me this error. So, I copied the OS Image on my user home folder. Closed the app and opened it again, selected the OS Image from the new location and everything worked fine and I was able to select my SD card as the target.

I hope this helps

adamburmister, 2025-05-05:
@yousihy that worked for me too.

chestnutsoftech, 2025-10-19:
As soon as I changed it t 32 gb usb 2.0 pendrive it worked. so problem is might be with usb 3.0 or 64gb. but to use that drive. I copied all contents to 64gb from 32gb. it started to as source drive. so I believe the issue is with the contents it has. if empty drive found it will go crazy. it needs folders from android.