balena-io/etcher #4052: Implement proper privilege elevation on macOS

elysweyr, 2023-04-17:
- **Etcher version:** any
- **Operating system and architecture:** macOS / any
- **What do you think should have happened:** Application should spawn new privileged process, elevation requests for spawned is accepted, privileged helper process performs flashing, communication to main process takes place via NSXPC (see https://developer.apple.com/documentation/foundation/nsxpcconnection for more info) 
- **What happened:** Etcher is creating a fake elevation prompt and processing the password on its own. See https://github.com/balena-io/etcher/blob/1a368f55fa023599310f6cc09a269751f99aea76/lib/shared/catalina-sudo/sudo-askpass.osascript-en.js#L8

aethernet, 2023-04-25:
Agree 100% with you that the current method is suboptimal (at best).

> Etcher is creating a fake elevation prompt
It's not a "fake" one, it's very real and totally working.
This predates better integrated os mechanism.

PR welcomes :)

dfunckt, 2023-07-12:
Closing in favor of #3321

closed 2023-07-12 by dfunckt