balena-io/etcher #4054: Lost Source and target drive after computer goes to sleep

ReinerKrause, 2023-04-20:
- **Etcher version: 1.5.109** 
- **Operating system and architecture: Win11 64bit** 
- **Image flashed: M2 SSD with USB adapter to USB Disk** 
- **What do you think should have happened: 
- Selected Source and target
- After that i started cloning which runs normally
- I've expected that the proces runs untile the end.
- **What happened:**
- The computer (Laptop connected to Power) goes into StandBy after 40% of the copy process
- After waking up the laptop, Balena Etcher remains blank and did not recover** 
- Both Disks remain disconnected from windows. Even after reboot the target (and especially) the source are no more visible in Windows Explorer.
- Also in disk management it is not possible to assign a drive letter.

--> Is there a way to get the Source drive back to work?

- **Do you see any meaningful error information in the DevTools? ** 
No informations!

aethernet, 2023-04-25:
Did you plug it out and back in? 

Etcher is only reading the source, there's no write action on it, whatever windows does to a disk when going to sleep is not controlled by etcher.

closed 2023-04-25 by aethernet

ReinerKrause, 2023-04-25:
That doesn't seem quite right to me.As soon as I connect the two drives to the computer, they get a drive letter.When I start the clone process, both drives are unmounted from the file system.This is the state that both the target and the source are now in after Balena Etcher crashed due to the standby.--

Closed #4054 as completed.