balena-io/etcher #4057: DEB repo: package list refresh fails, error 402 (payment required) apr 2023 edition

galagithub, 2023-04-26:
```
E: Failed to fetch https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu/dists/jammy/InRelease  402  Payment Required [IP: 54.192.235.42 443]
E: The repository 'https://dl.cloudsmith.io/public/balena/etcher/deb/ubuntu jammy InRelease' is no longer signed.
N: Updating from such a repository can't be done securely, and is therefore disabled by default.
N: See apt-secure(8) manpage for repository creation and user configuration details.
```

Just loose them! it's a re-occurring issue.

galagithub, 2023-04-26:
#4034 and #3511 say the same thing at different times

aethernet, 2023-04-27:
Hello,

We're moving out of Cloudsmith as continuing to host the package there would be too costly for us.

We're very thankful for the service and help CloudSmith provided, and would recommend their product for smaller open source app and private packages anytime!

At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self hosted (probably CloudFlare R2 + workers) in the upcoming weeks.

So sorry for the inconvenience and thanks you for your patience.

Note that you can always get a .deb and .rpm package in the release section of the github repository.

closed 2023-04-27 by aethernet