balena-io/etcher #4059: .deb, .rpm, apt, yum, 402 issue

aethernet, 2023-04-27:
Hello,

We're moving out of Cloudsmith as continuing to host the package there would be too costly for us.

We're very thankful for the service and help CloudSmith provided, and would recommend their product for smaller open-source apps and private packages anytime!

At the moment we don't have the bandwidth (pun intended) to set up an alternative repository but might be moving to self-hosted (probably CloudFlare) in the upcoming weeks.

So sorry for the inconvenience and thank you for your patience.

Note that you can always get a .deb and .rpm package in the release section of the GitHub repository.

lurch, 2023-04-27:
(It might also be worth providing instructions in your post above for how people can update their apt / yum configs to remove Cloudsmith, so that people stop getting this 402 error every time they do `sudo apt update`?)

wboswall, 2023-04-27:
Please provide an alternative way to install and update the package as this software is really awesome and would highly recommend it.

lurch, 2023-04-27:
> Note that you can always get a .deb and .rpm package in the release section of the GitHub repository.

i.e. https://github.com/balena-io/etcher/releases for those people not familiar with GitHub.

pmo-19, 2023-04-29:
Nope ! in my Ubuntu Focal 20.04 I have package balena-etcher-electron installed. It's no where here in  the releases ...
How to you build this package then ?

lurch, 2023-11-30:
> We're moving out of Cloudsmith...

Does that mean that the "OSS hosting by cloudsmith" button ought to be removed from https://etcher.balena.io/#download-etcher ?
[image]

peternewman, 2024-05-29:
You could always just use GitHub hosting and you can host a APT repo fine from there, there are even a few GitHub actions to auto-generate them. E.g. my branch of this one:
https://github.com/jrandiny/apt-repo-action/pull/1

KaKi87, 2024-09-02:
Hi,

A few months ago, I started developing a tool named [dynapt](https://git.kaki87.net/KaKi87/dynapt) to automatically fetch updates through APT for packages that don't provide an APT repository.

As I've been upgrading Etcher through it for quite some time now, I just published my first pre-release of it, which you can download and set up from [this tutorial](https://gist.github.com/KaKi87/f620788e9901abbfef4978eb7ad358b4).

For Etcher specifically, replace the following configuration line :

```json
    "apps": []
```

With the following :

```json
    "apps": [
        {
            "name": "Etcher",
            "github": {
                "repo": "balena-io/etcher"
            }
        }
    ]
```

For questions or issues, please submit a comment on the tutorial page.

Thanks

---

cc. @InklingGirl (#4181)

InklingGirl, 2024-09-02:
Oh thanks @KaKi87 it sounds promising, I'll look into it.