balena-io/etcher #4064: Flashing xz compressed file from URL fails

agners, 2023-05-11:
- **Etcher version:** 1.18.4-x64
- **Operating system and architecture:** Arch Linux x86-64
- **Image flashed:** https://github.com/home-assistant/operating-system/releases/download/10.1/haos_odroid-n2-10.1.img.xz
- **What do you think should have happened:** Flash that image
- **What happened:** Error: The elevated process died unexpectedly
- **Do you see any meaningful error information in the DevTools?** 

```
gui.js:37 Error: The elevated process died unexpectedly
    at createError (gui.js:37:219904)
    at Object.createUserError (gui.js:37:220117)
    at gui.js:341:5180
    at async withTmpFile (gui.js:1:718820)
    at async elevateCommand (gui.js:341:4411)
    at async Server.<anonymous> (gui.js:341:8515)
```

[image]

agners, 2023-05-11:
Same with the latest preview v1.18.8. An older version I had at hand seems to work (1.5.120).

sweavo, 2023-05-25:
samesame. thanks for the tip about 1.1.0.x

sweavo, 2023-05-25:
and 1.10x stalled out at 99% with 20seconds remaining.

dfunckt, 2023-07-13:
Does it work if you download the file first and try to flash that?

c0ffeeca7, 2023-07-13:
works for me:

**Etcher version**: 1.18.4-x64
**Operating system and architecture**: Ubuntu x86-64
**Image flashed**: https://github.com/home-assistant/operating-system/releases/download/10.1/haos_odroid-n2-10.1.img.xz, downloaded locally, not flashed from URL

agners, 2023-08-18:
@dfunckt yes it works when downloading and then installing. We currently document this as work around.

We've discovered that the same is true for MacOS.

Any chance that this gets addressed? It would be really helpful if we could use flashing directly from an URL again.

cmccall-epilog, 2024-03-07:
I'm also unable to flash from URL, but I don't get an error at all. The progress on bytes written increases until it just about completes and then it stops indefinitely. I can download the image and flash fine, but the workflow problem I'm trying to solve centers around eliminating the download step.

I'm on macOS 14.0

lukechilds, 2024-04-28:
I get the same issue when flashing `.img.gz` and `.img.xz` from a URL but flashing `.img.zip` works from a URL.

Flashing all three of the above compressed images works when flashing from the locally downloaded file.

lukechilds, 2024-04-28:
@dfunckt let me know if there's any more information I can provide to help you guys. We reccomend balenaEtcher to all of our users when installing @getumbrel and currently it's slowing down our build process quite a lot because we can't make use of parallel gzip compression on our images and instead use single threaded zip compression to avoid hitting this bug.

ShaharHD, 2024-06-01:
I have the same issue with `xz` files (not URL)
Happens on `1.19.21` and `1.19.19`
No issue when using the uncompressed version

I don't know at version it stopped working exacly, but used to work before.
I'm on macOS Sonoma 14.5 running on M1 Max

lanzer-gaia, 2024-08-20:
For anyone who came across this issue.  On MacOS the default unarchiver can decompress an .xz image, while on Windows you can use 7zip archiver.  As mentioned above, writing an uncompressed image will not show any issue.  That's one way to work around this error until a fix arrives.

lonely-soul1234, 2025-06-08:
used balena etcher linux x64 on ubuntu 24.04.2 LTS to flash a 128GB class 10 TF card from sandisk , the image flashed is ubuntu 24.04 orange pi from joshua-riek. the img.xz failed to be verified multiple times, but I unxz it using defualt tool and flashed properly.
Samething error happened writing the image to TF card (128gb or 32gb) on a Win 10 with balena etcher 1.19.25, either with or without admin privilige.  .xz file failed to flash, but extracted .img works fine.

I guess something is wrong with processing of .xz format. Next time I experience this issue, I will uncompress the .xz file and flash with .img file.