balena-io/etcher #4074: stuck @ 29% in Ubuntu live on HP T620 devide

dsimons2710, 2023-06-16:
Hi Alle,i try to flash a M2 ssd on my HP T620 client. I use a Ubuntu Live USB boot device to boot up the client, then i start firefox anddownload Baina Etcher to flash Home assistant to my m2 ssd thats in the client.Etcher start flashing but after 29% it stops and my system get stuck....only a reboot brings it back to live. someone any idea wats i can do to fix this issue?

dsimons2710, 2023-06-16:
when i try again, t again stops at 29%... every singel time......

andrewler, 2023-07-11:
Same issue on Dell Optiplex 7040m + Samsung 4k TV/monitor
Ubuntu 22.04 Live USB
Writing process hangs at 30% for a few minutes.
After that the Ubuntu frontend dies and restarts.

The writing process completes successfully when I reduce the HDMI output resolution from 4k to 1080p.