balena-io/etcher #4077: Link to Help Files Broken

spflanze, 2023-06-18:
- **Etcher version:** 
1.18.4

- **Operating system and architecture:** Ubuntu 22.04, 
 CPU: AMD Ryzen 9 5950X 16-Core Processor (3400 MHz)
Memory: 128724 MB
Concurrency: 32
OS Version: Linux 5.17.0-1020-oem #21-Ubuntu SMP PREEMPT Fri Oct 14 

- **Image flashed:**
NA

- **What do you think should have happened:** 
When the round icon with the "?" character in it is clicked on a help file should have opened.

- **What happened:**
A webpage opens with the message:
404 - page not found
The master branch of etcher does not contain the path  SUPPORT.md.

- **Do you see any meaningful error information in the DevTools?** 
No

lurch, 2023-06-20:
See #4056 ?

dfunckt, 2023-06-20:
Published [v1.18.8](https://github.com/balena-io/etcher/releases/tag/v1.18.8) as latest release with includes the fix.

closed 2023-06-20 by dfunckt