balena-io/etcher #4078: Flash drive dissapears after flashing via BalenaEtcher

winguy452, 2023-06-23:
- **Etcher version: 1.18.8** 
- **Windows 11 22H2** 
- **Image flashed: Olarila High Serria** 
- **What happened: I flashed a High Serrria .raw file, but when it was done, it couldn't be seen on Explorer! It also wasnt visible on diskpart either.**
- **Do you see any meaningful error information in the DevTools?** Nope. None

Please help me, I have been searching for days!

dfunckt, 2023-07-12:
The image likely does not contain partitions in a filesystem supported by Windows. Try inspecting the drive in a Linux-y OS.

closed 2023-07-12 by dfunckt

winguy452, 2023-07-13:
Ok thank you, I will try that.

winguy452, 2023-07-13:
> The image likely does not contain partitions in a filesystem supported by Windows. Try inspecting the drive in a Linux-y OS.

Ok thqnk you, i will try that.